roleoroleo/sonoff-hack #119: Firmware works, connecting to home assistant fails

andregu, 2022-11-04:
i have a 

Hostname | sonoff-88d3
Firmware Version | 0.1.4 (made the log fix change)
Base Version | V5520.2053.0411build20220811
Model | GK-200MP2-B
Device ID | a63001d07c

install the sonoff-hack no problem, super eays, using VLC to stream [rtsp] works fine, when i try to add it to home assistant (i have MQTT enabled and connecting to the broker) either via the Yi component, or the ONVIF integration , it never connects,

 not sure what to do anymore

roleoroleo, 2022-11-05:
Which integration are you using?

darkxst, 2022-11-05:
This release added MQTT autodiscovery for Home Assistant. Provided MQTT is all configured Home Assistatant should automatically detect the device and add motion and motion_image sensors (no need for the Yi component unless you want some of the other features it provides).  They may not appear in the dashboard automatically though. 

Can you see the camera under Settings -> Integrations -> MQTT -> devices?

Can you see messages from the camera in MQTT explorer? There should messages under the following topics:
`homeassistant/binary_sensor/<device_id>_motion/config`
`sonoffcam/<device_id>`

andregu, 2022-11-05:
> Which Yi integration are you using?

https://github.com/roleoroleo/yi-hack_ha_integration/ this one, so yours i guess

> This release added MQTT autodiscovery for Home Assistant. Provided MQTT is all configured Home Assistatant should automatically detect the device and add motion and motion_image sensors (no need for the Yi component unless you want some of the other features it provides). They may not appear in the dashboard automatically though.
> 
> Can you see the camera under Settings -> Integrations -> MQTT -> devices?
> 
> Can you see messages from the camera in MQTT explorer? There should messages under the following topics: `homeassistant/binary_sensor/<device_id>_motion/config` `sonoffcam/<device_id>`
> 
> EDIT: are you using wireless or ethernet? Onvif by default listens on Ethernet port. If on wireless change that setting to Wireless

i can see the camera under Settings -> Integrations -> MQTT -> devices

i can see those topics existing with MQTT explorer but i dont see many messages flowing

i have changed it to wireless since that is what i am using at the moment

andregu, 2022-11-05:
[config (1).tar.gz](https://github.com/roleoroleo/sonoff-hack/files/9943778/config.1.tar.gz)
 if it helps here is my config

roleoroleo, 2022-11-05:
Check if you are able to see something with a mqtt client (MQTT Explorer).

darkxst, 2022-11-05:
> i can see those topics existing with MQTT explorer but i dont see many messages flowing, and the card in the dashboard has no image and the motion always says unknown

For your model camera you need to set `Motion Event Location` = `0` on the MQTT Settings page. That is the last setting on that page.

For some reason this setting is empty in your config file.

andregu, 2022-11-06:
> For your model camera you need to set `Motion Event Location` = `0` on the MQTT Settings page. That is the last setting on that page.
>
> For some reason this setting is empty in your config file.

> Check if you are able to see something with a mqtt client (MQTT Explorer).

with `Motion Event Location` = `0` now i can see the messages in MQTT explorer and the card in the HA dashboard works, but the YI integration or the ONVIF still dont, when i try the YI integration just gives me timeout (i disabled the cloud not sure it matters) and ONVIF gives me this error in the HA logs

```
2022-11-06 11:16:40.966 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/httpcore/backends/asyncio.py", line 33, in read
return await self._stream.receive(max_bytes=max_bytes)
File "/usr/local/lib/python3.10/site-packages/anyio/_backends/_asyncio.py", line 1265, in receive
await self._protocol.read_event.wait()
File "/usr/local/lib/python3.10/asyncio/locks.py", line 214, in wait
await fut
asyncio.exceptions.CancelledError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/httpcore/_exceptions.py", line 8, in map_exceptions
yield
File "/usr/local/lib/python3.10/site-packages/httpcore/backends/asyncio.py", line 31, in read
with anyio.fail_after(timeout):
File "/usr/local/lib/python3.10/site-packages/anyio/_core/_tasks.py", line 118, in __exit__
raise TimeoutError
TimeoutError
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/httpx/_transports/default.py", line 60, in map_httpcore_exceptions
yield
File "/usr/local/lib/python3.10/site-packages/httpx/_transports/default.py", line 353, in handle_async_request
resp = await self._pool.handle_async_request(req)
File "/usr/local/lib/python3.10/site-packages/httpcore/_async/connection_pool.py", line 253, in handle_async_request
raise exc
File "/usr/local/lib/python3.10/site-packages/httpcore/_async/connection_pool.py", line 237, in handle_async_request
response = await connection.handle_async_request(request)
File "/usr/local/lib/python3.10/site-packages/httpcore/_async/connection.py", line 90, in handle_async_request
return await self._connection.handle_async_request(request)
File "/usr/local/lib/python3.10/site-packages/httpcore/_async/http11.py", line 105, in handle_async_request
raise exc
File "/usr/local/lib/python3.10/site-packages/httpcore/_async/http11.py", line 84, in handle_async_request
) = await self._receive_response_headers(**kwargs)
File "/usr/local/lib/python3.10/site-packages/httpcore/_async/http11.py", line 148, in _receive_response_headers
event = await self._receive_event(timeout=timeout)
File "/usr/local/lib/python3.10/site-packages/httpcore/_async/http11.py", line 177, in _receive_event
data = await self._network_stream.read(
File "/usr/local/lib/python3.10/site-packages/httpcore/backends/asyncio.py", line 30, in read
with map_exceptions(exc_map):
File "/usr/local/lib/python3.10/contextlib.py", line 153, in __exit__
self.gen.throw(typ, value, traceback)
File "/usr/local/lib/python3.10/site-packages/httpcore/_exceptions.py", line 12, in map_exceptions
raise to_exc(exc)
httpcore.ReadTimeout
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
resp = await request_handler(request)
File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
resp = await handler(request)
File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 222, in forwarded_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
return await handler(request)
File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 82, in ban_middleware
return await handler(request)
[... 24 lines cut ...]
File "/usr/local/lib/python3.10/site-packages/zeep/transports.py", line 230, in post_xml
response = await self.post(address, message, headers)
File "/usr/local/lib/python3.10/site-packages/zeep/transports.py", line 215, in post
response = await self.client.post(
File "/usr/local/lib/python3.10/site-packages/httpx/_client.py", line 1842, in post
return await self.request(
File "/usr/local/lib/python3.10/site-packages/httpx/_client.py", line 1527, in request
return await self.send(request, auth=auth, follow_redirects=follow_redirects)
File "/usr/local/lib/python3.10/site-packages/httpx/_client.py", line 1614, in send
response = await self._send_handling_auth(
File "/usr/local/lib/python3.10/site-packages/httpx/_client.py", line 1642, in _send_handling_auth
response = await self._send_handling_redirects(
File "/usr/local/lib/python3.10/site-packages/httpx/_client.py", line 1679, in _send_handling_redirects
response = await self._send_single_request(request)
File "/usr/local/lib/python3.10/site-packages/httpx/_client.py", line 1716, in _send_single_request
response = await transport.handle_async_request(request)
File "/usr/local/lib/python3.10/site-packages/httpx/_transports/default.py", line 352, in handle_async_request
with map_httpcore_exceptions():
File "/usr/local/lib/python3.10/contextlib.py", line 153, in __exit__
self.gen.throw(typ, value, traceback)
File "/usr/local/lib/python3.10/site-packages/httpx/_transports/default.py", line 77, in map_httpcore_exceptions
raise mapped_exc(message) from exc
httpx.ReadTimeout

```

andregu, 2022-11-06:
ill give some more information, i downloaded ONVIF device manager for windows, it auto detects my camera, all information is there, PTZ works, streaming does not

darkxst, 2022-11-06:
> ill give some more information, i downloaded ONVIF device manager for windows, it auto detects my camera, all information is there, PTZ works, and streaming works too

If ONVIF works there it should also work in HA.

Both the Yi integration and onvif rely on HTTP connections to the camera, but otherwise are completely independent of each other. Perhaps your HA box has an issue or is unable to connect to the camera for some reason. 

Perhaps try the generic camera integration setting the snapshot url and rtsp stream. See below link. Can that connect?
https://www.home-assistant.io/integrations/generic/

andregu, 2022-11-06:
it does not, would it matter that my HA is running in kubernetes? I have seen some stuff saying that since i only have 1 port forward it cant do RTSP

andregu, 2022-11-06:
switched to host mode in my kubernetes configuration and it seems to be working now

darkxst, 2022-11-06:
> it does not, would it matter that my HA is running in kubernetes? I have seen some stuff saying that since i only have 1 port forward it cant do RTSP

Depending on how its configured yes that will matter. I am not really familiar with Kubernetes, however if HA is running on an internal network in the container and then you are using a port forward to access HA, thats probably not going to work. 

The standard HA Docker setup binds the LAN into the container, so that HA is just another host on your LAN.

roleoroleo, 2022-11-06:
If you (or HA) uses discovery protocol (wsdd), ha and the cam must be on the same network subnet.
Otherwise, you can set manually the onvif ip:port. Onvif port is 1000.
This last solution should work with kubernetes in internal network.

closed 2024-05-04 by github-actions[bot]