roleoroleo/sonoff-hack #120: Wiki update for 0.1.4

darkxst, 2022-11-05:
Hey Roleo,
  Can you update the wiki for the new release. I left in the manual config snippets mainly for reference in case people still want to setup manual sensors with different settings (probably only needed if they disable discovery!).

-----------------------------------------------------------------------------------
Event detection
The only feature not present in ONVIF is the event detection. You can use the MQTT sensors for this purpose.

Release 0.1.4 introduced MQTT auto-discovery for Home Assistant, so the MQTT sensors will be available without any configuration of yaml files. You just need to enable mqtt on the camera.  If the sensors don't appear on your dashboard you can find the camera under Settings -> devices and add them from there.

Access to the web interface
Go to MQTT section
Enable mqtt and set the server parameters (ip, port, authentication, etc...)
Optionally set a name for the device under Home Assistant Discovery -> Name
For motion events to work on GK-200MP2-B and S-CAM cameras set `Motion event location = 0`
All other settings are optional
Save configuration and reboot

You can also manually configure sensors with the following yaml code. You probably want to configure a custom Topic prefix in the web interface if using these.

Motion Sensor:

mqtt:
  binary_sensor:
    - state_topic: "home-assistant/kitchen/motion_detection"
      name: "Motion detection - kitchen"
      payload_on: "motion_start"
      payload_off: "motion_stop"
      qos: 1

Camera that displays screenshots during a motion detection event:

mqtt:
  camera:
    - topic: "home-assistant/kitchen/motion_detection_image"
      name: "Motion detection image - kitchen"

darkxst, 2022-11-05:
Here is a screenshot of the detected camera in Home Assistant. Could maybe include that also?
[image]

darkxst, 2022-11-05:
Screenshot of mqtt discovered device in HA if you want to include that.
[image]

roleoroleo, 2022-11-05:
Wiki updated.
Thank you.

darkxst, 2022-11-05:
Slightly tweaked this sentence now its not a list!

------
Access the web interface an go to the MQTT section, enable mqtt and set the server parameters (ip, port, authentication, etc...). Optionally set a name for the device under Home Assistant Discovery -> Name. For motion events to work on GK-200MP2-B and S-CAM cameras set Motion event location = 0. All other settings are optional, save configuration and reboot camera.

closed 2022-11-07 by darkxst