roleoroleo/sonoff-hack #122: PTZ not working on 0.1.4

IAmTheMilkManMyMilkIsDelicious, 2022-12-24:
Firmware Version | 0.1.4
-- | --
Base Version | V2524.1.245build20191030
Model | GK-200MP2B

Since upgrading from 0.1.2 to 0.1.4 the ptz doesn't function, specifically the web ui ptz, I don't use ONVIF.
Clicking the button in the UI doesn't make the camera move but sends a get request that returns:
``` json
{
    "error":"false"
}
```

I ssh-ed into the device and manually ran the ptz command and it fails there. Never seen this error before, nothing on the internet about it, hopefully someone here has seen it.

``` sh
$ ptz -a right
ptz: can't resolve symbol 'hw_ptz_sendptz'
```

Thanks

roleoroleo, 2022-12-25:
Check here:
https://github.com/roleoroleo/sonoff-hack/issues/93#issuecomment-1351560793

IAmTheMilkManMyMilkIsDelicious, 2022-12-25:
Awesome that cp command fixed it thanks, I didn't think the issue affected the GK-200MP2B

closed 2022-12-25 by IAmTheMilkManMyMilkIsDelicious