roleoroleo/sonoff-hack #123: GK-200MP2-B hack reports as GK-200MP2C

ggggh, 2023-01-04:
Loaded GK-200MP2-B on SD card (as that is what my camera seems to be), confirmed the sonoff-hack/model file states the same, but on the sonoff hack web portal, once camera is up, it says GK-200MP2C.  Is that a bug on the portal, is it actually detecting it is a model C (regardless of what the packaging says, and therefore I loaded the wrong firmware)?

I'm having a lot of trouble getting some things to work (motion alerts are triggered but files are only saved in /tmp, not in /mnt/mmc/alarm_record/, and I can't seem to get the "motion event location" setting to actually work)

roleoroleo, 2023-01-06:
Where do you see the model GK-200MP2C?

ggggh, 2023-01-06:
On the sonoff hack status page (first tab):

`Firmware Version	0.1.4
Base Version	V0525.1.276build20201103
Model	GK-200MP2C`

roleoroleo, 2023-01-07:
The status page reads this value from a file of the original fw, so it can't be wrong.
From https://github.com/roleoroleo/sonoff-hack/blob/master/src/www/httpd/cgi-bin/status.json:
```
MODEL_CFG_FILE=/mnt/mtd/ipc/cfg/config_cst.cfg
[ -f /mnt/mtd/db/conf/config_cst.ini ] && MODEL_CFG_FILE=/mnt/mtd/db/conf/config_cst.ini
...
MODEL=$(cat $MODEL_CFG_FILE | grep model | cut -d'=' -f2 | cut -d'"' -f2)
```
Are you sure that you used the correct hack file?

roleoroleo, 2023-01-07:
The model you find in /mnt/mmc/sonoff-hack/model comes from the hack, so it depends on the hack file you downloaded.
/mnt/mtd/ipc/cfg/config_cst.cfg is a sonoff file.

Probably your cam is a GK-200MP2C.

ggggh, 2023-01-07:
Ok! Is there a way to tell from the camera itself? Any outside differences? The box it came in clearly says MP2-B

roleoroleo, 2023-01-08:
The truth is the file config_cst.cfg..
I don't know if there are exterior differences, I only own the old model GK-200MP2B.
What's the sonoff fw version?

ggggh, 2023-01-08:
Do I need to check on the ewelink app for the original sonoff fw version?
PTZ was working but extremely slow.
What wasn't working was motion events. They're triggered, but files are only saved in /tmp, not in /mnt/mmc/alarm_record/, and I can't seem to get the "motion event location" setting to actually work.
I'll load the C hack and try it out.

ggggh, 2023-01-08:
Changed to C version of the hack, but still the same issue on /mnt/mmc/alarm_record/ being empty and consequently no FTP uploads, although video files for the events do pile up in /tmp.  Any ideas?

roleoroleo, 2023-01-09:
Your fw version is V0525.1.276build20201103 if I'm not wrong.

> What wasn't working was motion events. They're triggered, but files are only saved in /tmp, not in /mnt/mmc/alarm_record/

This is strange: /tmp is a temporary filesystem. What happens when you restart the cam?
Is your sd card properly formatted?

ggggh, 2023-01-10:
Files appear with a "/tmp/file*****" format, with the * being random alphanumeric.  They do disappear after a reset, as do the log files in /tmp.
Interestingly, the video files used to pile up on /tmp, but right now they're disappearing in seconds from /tmp, although they still don't make it to /mnt/mmc/alarm_record.  Not sure where they're going.
Neither ftppush nor thumb scripts are doing that, as they only look in alarm_record.

MQTT messages are being sent on each motion event, but the hack portal is not listing the events in the motion events table.
Sqlite shows them on t_alarm_log, but there is nothing on T_RecordFile.
Motion Event setting is on 1;  changing it to 0 only seems to cause the mqtt messages to not be sent, but otherwise stays the same (/tmp/file is still created and deleted quickly).

roleoroleo, 2023-01-16:
> Files appear with a "/tmp/file*****" format, with the * being random alphanumeric. They do disappear after a reset, as do the log files in /tmp.

They are temporary files, you shouldn't pay attention to these.
The problemi is: why aren't they moved to the alarm directory?
Sonoff firmware is responsible fot this action (save mp4 files).
If you remove the sd card and disable the hack, are the files saved?
Have you already tried using another sd card?

> MQTT messages are being sent on each motion event, but the hack portal is not listing the events in the motion events table.
> Sqlite shows them on t_alarm_log, but there is nothing on T_RecordFile.

It depends on the Sonoff fw version: the old firmware reads events from /mnt/mtd/db/ipcsys.db (table t_alarm_log), the new one from /mnt/mmc/AVRecordFile.db (table T_RecordFile).
If your T_RecordFile is empty and mqtt messages are sent, "Motion Event" setting must be 1.

korskantevin, 2023-02-02:
I've got GK-200MP2-B written on the box, but GK-200MP2С model written in Ewelink app.

korskantevin, 2023-02-02:
GK-200MP2-B is on the bottom of camera and on box but GK-200MP2С model written in Ewelink app. what file should I use? I've install hack for GK-200MP2С model yesterday

roleoroleo, 2023-02-02:
What's the firmware version reported by the app?

korskantevin, 2023-02-02:
<html>
<body>

Base Version | V0525.1.412build20220712
-- | --

</body>
</html>

roleoroleo, 2023-02-04:
I didn't test this version.
Please, send me a backup and I will check if it's possible to support it.

korskantevin, 2023-02-04:
> backup

what do you mean by backup? be more specific please :)

roleoroleo, 2023-02-05:
Check the wiki:
https://github.com/roleoroleo/sonoff-hack/wiki/Dump-your-backup-firmware-(SD-card)

korskantevin, 2023-02-06:
> Check the wiki: https://github.com/roleoroleo/sonoff-hack/wiki/Dump-your-backup-firmware-(SD-card)

> I didn't test this version. Please, send me a backup and I will check if it's possible to support it.

https://www.dropbox.com/s/4wxf866986wx2kn/backup.zip?dl=0

jekkel, 2023-03-04:
FTR Same story here:
* bought as `GK-200MP2-B`, labelled on the box and on the camera as `GK-200MP2-B`
* hack says:
  * Model: GK-200MP2C
  * Base Version: V0525.1.412build20220712

Running hack v0.1.4 with working PTZ (albeit slow), not sure about motion detection as I don't really use it (I'd say it doesn't work properly).

Worth noting: upgrading to hack v0.1.5 makes it crash loop after about 5 minutes uptime, had to downgrade.

Anything I can help with?

rlarranaga, 2023-04-24:
Same with me. Trying to figure out which firmware to use?
GK-200MP2-B on the box, GK-200MP2C on ewelink
Thanks

korskantevin, 2023-04-24:
> Same with me. Trying to figure out which firmware to use? GK-200MP2-B on the box, GK-200MP2C on ewelink Thanks

I've try both of firmwares and finally chose GK-200MP2-B because PTZ was not working on GK-200MP2C.

rlarranaga, 2023-04-24:
Thanks for sharing your experience!
I am currently on that fw too.

roleoroleo, 2023-04-26:
Please, let me know if motion detection works and what "Motion event location" configuration.

korskantevin, 2023-05-02:
> Please, let me know if motion detection works and what "Motion event location" configuration.
Yes it works.
[image]
"Motion event location" where can i find this?

Buy the way. I set up MQTT and it works fine, but how can I stop motion records, and get only motion event by MQTT?

roleoroleo, 2023-05-04:
> "Motion event location" where can i find this?

I mean this option:
[image]

> Buy the way. I set up MQTT and it works fine, but how can I stop motion records, and get only motion event by MQTT?

I think it's not possible.

darkxst, 2023-05-05:
The recordings are made by the Sonoff software. If you dont want to keep them you could setup a cron job to delete them regularly.

korskantevin, 2023-05-05:
> The recordings are made by the Sonoff software. If you dont want to keep them you could setup a cron job to delete them regularly.

yes, I know that. but I'm worried about SD card, which I think will be corrupter after approximately one year

roleoroleo, 2023-05-05:
> yes, I know that. but I'm worried about SD card, which I think will be corrupter after approximately one year

Maybe 2 years, but you are right.
It depends on the model, but you could try to kill the process AVRecorder and check what happens.

korskantevin, 2023-05-05:
> try to kill the process AVRecorder and check what happens

records are still being saved. but MQTT stops send me alarms

closed 2024-05-04 by github-actions[bot]