roleoroleo/sonoff-hack #125: Static Ip

GeorgeGero, 2023-01-28:
This is a really nice project. Gongratulations for that ... 
I want to change the IP of the camera to static from DHCP because th Blue Iris machine is to another subnet . 
Is it possible to do this ?
Thanks in advance

roleoroleo, 2023-01-29:
I really don't know if it's possible.
You can try to set the ip address adding the file `/mnt/mmc/startup.sh`
with the line `ifconfig wlan0 YOUR.IP.ADD.ESS` if you are using wifi or ifconfig eth0 `YOUR.IP.ADD.ESS` if you are using ethernet.
But I don't know if there is some sonoff process that overwrites this setting.

GeorgeGero, 2023-01-30:
Thanks for your reply .. In the sd card I see a file boot.sh (not startup.sh) .. But I see that it calls a script (dhcp.sh) .. I will try to add the ifconfig line at the end ..This is the code inside the boot.sh

# Fix hostname
cp /mnt/mtd/ipc/app/script/dhcp.sh /tmp/dhcp.sh
HOSTNAME=$(cat /var/sdcard/sonoff-hack/etc/hostname)
sed -i "s/dhcpname=\$2/dhcpname=$HOSTNAME/g" /tmp/dhcp.sh
mount --bind /tmp/dhcp.sh /mnt/mtd/ipc/app/script/dhcp.sh

closed 2024-05-04 by github-actions[bot]