roleoroleo/sonoff-hack #126: What about Intercom functionality?

korskantevin, 2023-02-02:
i've got GK-200MP2-B written on the box, but GK-200MP2С model written in Ewelink app. What about Intercom functionality? Ewelink app support it, but hack does not.

darkxst, 2023-02-02:
>What about Intercom functionality? Ewelink app support it, but hack does not.

eWelink uses a proprietary API for two way audio. That is not possible to support.

korskantevin, 2023-02-02:
I mean create second incoming audio stream from Home Assistant or HomeBridge in my case to use it in  Apple Home app

darkxst, 2023-02-02:
Audio from the camera to Home Assistant should work with both RTSP and ONVIF. Its not possible to send audio back to the speaker in the camera though.

korskantevin, 2023-02-02:
sadly to hear it :( it is very useful feature

roleoroleo, 2023-02-02:
As darkst wrote, at the moment it's not supported.
I never tried to implement this feature.

korskantevin, 2023-02-02:
> As darkst wrote, at the moment it's not supported. I never tried to implement this feature.

can you try? please :)

darkxst, 2023-02-03:
> can you try? please :)

It doesnt make sense to add support to Sonoff-hack for the eWeLink API. It would need to be implemented in Home Assistant, probably via the Sonoff-Lan addon. That project already has some support for the eWeLink API's, including PTZ control. But does not currently have any support for camera video or audio streams.

Adding ONVIF Profile T with audio backchannel to sonoff-hack would be the right way to do it, as it would effectively bypass the proprietary API, however would be a significant chunk of work. It would also require direct access to the audio hardware, which I dont think we have, (RTSP is handled by Sonoff code). Also profile T is not yet supported in Home Assistant either afaik, but would work in apps like Onvier.

korskantevin, 2023-02-03:
1) I'm not talking about eWeLink API, only that they has this functionality.
2) And I dont know what protocol is used by HomeKit. But I know that HomeKit cameras has Intercom button in Apple Home App. I think it is not ONVIF. But it will be very usefull to add this feature to sonoff-hack to bring Intercom button to Apple Home App. 
3) Don't know about Home Assistant, but there is "Camera FFmpeg" plugin for HomeBridge and there is "Two-way Audio Target" param in each camera settings - "The FFmpeg output command for directing audio back to a two-way capable camera."

darkxst, 2023-02-03:
> * I'm not talking about eWeLink API, only that they has this functionality.

Being a Sonoff product, this camera is built upon that API. The two-way audio you see in the eWeLink App, is using this API. It is also currently the only known way to send audio to the camera.

> 3\. "Camera FFmpeg" plugin f

Looks like this is using the RTSP backchannel directly. That is the same mechanism that ONVIF uses under the hood. Sonoff's RTSP server doesnt support that.

osos, 2023-02-17:
The WebRTC integration for HA should now support 2-way audio for supported cameras (ONVIF Profile T I presume)
See: https://github.com/AlexxIT/WebRTC#stream-to-camera

closed 2024-05-04 by github-actions[bot]