roleoroleo/sonoff-hack #128: GK-200MP2-B New Firmware - not tested don't upgrade!

darkxst, 2023-02-02:
Looks like Sonoff is pushing a new firmware `50420230103` out for the GK-200MP2-B . Dont upgrade to this, if you can help it. It hasnt been tested with Sonoff-hack just yet.

vlaraort, 2023-02-11:
I can tell that I upgraded and rtsp is not working :( I can perform some tests if you want to try something

Mr-P-Butthole, 2023-03-11:
I can't get it to work as my unit took new firmware before applying this overlay. I will make a separate issue once I rule out the sd-card and all other factors.

XzaR90, 2023-09-15:
What is the progress of this? I got version 50520230320 looks like it is not working on that firmware.

roleoroleo, 2023-09-24:
Sorry but I can't help with this issue.
I don't own this model.

darkxst, 2023-09-24:
> I got version 50520230320

I have this build on one of my cameras and its working fine with Sonoff Hack. Probably best to open a new issue if you can't get it working. I am going to close this issue as its obsolete!

closed 2023-09-24 by darkxst