roleoroleo/sonoff-hack #129: Snapshot on ptz page makes CPU usage to 85%

korskantevin, 2023-02-03:
There is setTimeout function set to interval=1000 in all.js. in my case snapshot answer is only after 3 seconds. And "top" command via SSH says 85% user CPU usage. So I think more accurate is to call on next snapshot only after receiving previous or get an error of previous call

darkxst, 2023-02-04:
You could try something like this (not tested though):
```
        interval = 1000;

        (function p() {
            jQuery.get('cgi-bin/snapshot.sh?base64=yes', function(data) {
                image = document.getElementById('imgSnap');
                image.src = 'data:image/jpeg;base64,' + data;
            }).always(function() {
                setTimeout(p, interval);
           });
            
        })();
```

korskantevin, 2023-02-04:
Works great! Thank you!

closed 2023-02-04 by korskantevin

roleoroleo, 2023-02-04:
Added to the master.
Thank you.