roleoroleo/sonoff-hack #131: "motion event location" on GK-200MP2-B and v5520

josecarre, 2023-02-06:
Hi, when trying to set this field to `0` as suggested, its reverted once I reboot. Any way to fix it manually via CLi or something? Thanks

[image]

My camera:

[image]

darkxst, 2023-02-06:
You can set it here, but I guessing its missing the key as no value has been read in the web interface
`/mnt/mmc/sonoff-hack/etc/mqtt-sonoff.conf`

Did you upgrade to 0.1.4 from an older version (or was it a fresh install?)

josecarre, 2023-02-06:
Hi, I tested both ways, upgrade and then reinstall, both with same behavior, but when doing fresh install maybe some old files affected?

In any case, heres is the file... can I just edit it and reboot?

```
[root@sonoff-garage]# cat /mnt/mmc/sonoff-hack/etc/mqtt-sonoff.conf
###############################################################################
#                       Configuration file for mqtt-sonoff                    #
###############################################################################

# -----------------------------------------------------------------------------
# Basic MQTT settings
# -----------------------------------------------------------------------------

MQTT_IP=xxxxxxx
MQTT_PORT=1883

MQTT_USER=xxxxxxx
MQTT_PASSWORD=xxxxxxx

MQTT_PREFIX=sonoffcamgarage

# -----------------------------------------------------------------------------
# Set the topics where the messages will be published
# The final topic will be MQTT_PREFIX/TOPIC_MOTION
# -----------------------------------------------------------------------------

TOPIC_BIRTH_WILL=status
TOPIC_MOTION=motion_detection
TOPIC_MOTION_IMAGE=motion_detection_image
MOTION_IMAGE_DELAY=0.5

# -----------------------------------------------------------------------------
# Set the topics messages
# -----------------------------------------------------------------------------

BIRTH_MSG=online
WILL_MSG=offline
MOTION_START_MSG=motion_start
MOTION_STOP_MSG=motion_stop

# -----------------------------------------------------------------------------
# Home Assistant Discovery Settings
# -----------------------------------------------------------------------------

HA_DISCOVERY=1
HA_CONF_PREFIX=homeassistant
HA_NAME_PREFIX=sonoff-garage

# -----------------------------------------------------------------------------
# Other settings
# -----------------------------------------------------------------------------

MQTT_KEEPALIVE=120
MQTT_QOS=1
MQTT_RETAIN_BIRTH_WILL=1
MQTT_RETAIN_MOTION=0
MQTT_RETAIN_MOTION_IMAGE=0
MQTT_IPCSYS_DB=0[root@sonoff-garage]#
```

darkxst, 2023-02-06:
its already set to 0
`MQTT_IPCSYS_DB=0`
but missing a newline at the end, maybe that is affecting the value not being read by web interface.

both PREFIX values are optional, they use the cam unique id if not set.

josecarre, 2023-02-06:
What line it is missing? can I add it manually? motion stuff doesnt seems to work very good, maybe is related

darkxst, 2023-02-06:
there is a newline/carriage return missing after the 0, sure you can add that manually. however I dont think that would affect the mqtt stuff, maybe only the value displayed in the web.
`MQTT_IPCSYS_DB=0[root@sonoff-garage]#`

The motion events on these cameras do take about 40secs to trigger, as the database is not updated until the recording is finished.

josecarre, 2023-02-06:
40sec... ohhh... It's a pity this GK-200MP2-B are so bad, the GK-200MP2B that i have works pretty well but it seems not on sale anymore... the new one is... not very useful with that 40s thing :(, thanks anyway! i guess i'll need to use just the internal recordings as the more instant videos to check.

darkxst, 2023-02-06:
The real time detection is still there, but the Sonoff firmware changed how its handled.
It will come to sonoff-hack whenever I get around to tackling #113

darkxst, 2023-02-06:
Also there seem to be a few people getting the `GK-200MP2C` which is labelled incorrectly as  GK-200MP2-B #123 . That would presumably still have the original motion detection in place in the firmware.

josecarre, 2023-02-06:
> The real time detection is still there, but the Sonoff firmware changed how its handled. It will come to sonoff-hack whenever I get around to tackling #113

Good to hear that, thanks @darkxst !

closed 2024-05-04 by github-actions[bot]