roleoroleo/sonoff-hack #133: HA discovery needs update for MQTT config options

darkxst, 2023-02-18:
Now that MQTT config options are there we should expose these to Home Assistant in the discovery config messages.

Not sure if the fact these are in a different process, will cause any issues with this. However i do notice that there are now two different versions of the config handling code, with the version in mqtt-config pre-dating the refactoring I did in mqtt-sonoff?

Also not sure, but some of the settings may require reboot of camera to take effect?

roleoroleo, 2023-02-27:
Please, explain me better.
I don't understand exactly what you mean.

darkxst, 2023-02-27:
Need to create the json config messages for each new mqtt endpoint, for HA discovery to work. Like what is done here for the motion and image sensors.

https://github.com/roleoroleo/sonoff-hack/blob/master/src/mqtt/mqtt-sonoff/src/mqtt-sonoff.c#L416

There is also a lot of duplicated code between mqtt-sonoff and mqtt-config. Not really sure why they need to be separate? I would be inclined to merge the config stuff into the original mqtt module.

Most changes to the sql database only seem to get read on a reboot, unless we can somehow force reloading of the sql database, by restarting some process.

roleoroleo, 2023-03-06:
You are right.
There is confusion between mqtt-sonoff and mqtt.
The reason is that they are partly inherited from yi cams where the behavior is different.
In those projects I have to keep the roles separate: that's why they are 2 programs.
But here, in this project, it's not necessary. We could do some refactoring work and improve the code.
We also need to bring together the messages used by yi-hack-integration and the messages used for HA discovery (and HA discovery is not my strong suit).

> Most changes to the sql database only seem to get read on a reboot, unless we can somehow force reloading of the sql database, by restarting some process.

I'm studying this problem, but no luck at the moment.

darkxst, 2023-03-07:
> We also need to bring together the messages used by yi-hack-integration and the messages used for HA discovery (and HA discovery is not my strong suit).

The messages used for HA discovery are JSON snippets that tell Home Assistant how to configure the sensor for a given MQTT endpoint.   They sit in a different namespace and don't have any effect on the underlying MQTT endpoints providing the data.  So are seperate from what yi-hack-integration uses.

roleoroleo, 2023-03-07:
Ok, I will work on this refactor.

roleoroleo, 2023-03-11:
This is the 1st step:
https://github.com/roleoroleo/sonoff-hack/commit/a08e4c889da022ac35cae4b9660668d56e0be6a5

Now I have to fix ha discovery.

closed 2024-05-04 by github-actions[bot]