roleoroleo/sonoff-hack #136: It control not working

gmcaesar, 2023-02-19:
Can not turn on or off or lights from web or HA

roleoroleo, 2023-02-19:
Please, explain me better.

darkxst, 2023-02-21:
I never got the IR led to work in my testing previously (directly via SQL).

However I did find that can control the visible (Status) LED's with the following (empty) file
`/dayun/mtd/cfg/disable_ledshow`

Might be worth adding that to the Web UI

gmcaesar, 2023-02-21:
I see the IR controls in the web UI but they seem to have no effect, I they always are in auto mode no matter what I do in the settings

darkxst, 2023-02-22:
@gmcaesar What camera model do you have? I just tested on my GK-200MP2-B and IR lamp control is working via Web UI.  What I described earlier was probably tested on the S-CAM model, which has the status LED on the face of the camera.

gmcaesar, 2023-02-22:
I have the same as you the Sonoff B

gewinh, 2023-02-27:
The IR control on the webui does not seem to have any effect on the GK-200MP2C. If I change the setting to off on the webui and do a save it reverts back to auto. If I reboot the camera it always comes back set to auto. Base Version V0525.1.412build20220712, firmware 0.1.5. The IR lamps stay on regardless. It would be nice to be  able to turn them off, one of my cameras faces a window and the IR lamps just reflect back.

gewinh, 2023-02-27:
I gave that a try using the webui, going from auto to off. I did get the ui to save the off state, but after a reboot of the camera it reverted back to auto. I tried this 3 times, each time the state was saved as off but after a reboot it reverted to auto.

roleoroleo, 2023-02-28:
Ok, try to check if the param is saved running this statement:
`# sqlite3 /mnt/mtd/db/ipcsys.db "select * from t_sys_param where c_param_name=\"InfraredLamp\";"`
Both afer the command and after a reboot.

gewinh, 2023-03-01:
Thanks for you ongoing support. I tried the database query. The reply was a value of 2 for the InfraredLamp, the value of 2 was always the same regardless of how many times I tried to change the IR LED setting, whether I saved it,  rebooted after saving, refreshing the web page etc. I thought it might be related to the web upgrade from 0.1.4 to 0.1.5 so I reflashed the SD from scratch with 0.1.5  and reconfigured it. Trying the test to change the IR lamp gave the same result ie 2  from the database. I thought the camera might be too busy during the test to write a new value to the database, so I set privacy on and disabled ftp, then did the test again and got the same result.

roleoroleo, 2023-03-01:
2 means "auto".
Please, try these new files:
[cgi.tgz](https://github.com/roleoroleo/sonoff-hack/files/10858509/cgi.tgz)

gewinh, 2023-03-01:
Thanks I will give it a try,  I did run the command sqlite3 /mnt/mtd/db/ipcsys.db "select * from t_sys_param where c_param_name=\"InfraredLamp\";"  from you camera_settings.sh and it did update the db so when I used your query above it results in a value of 1. Even after a reboot the value is 1, although the webui still says auto.

gewinh, 2023-03-01:
OK so I used the command sqlite3 /mnt/mtd/db/ipcsys.db "update t_sys_param set c_param_value=2 where c_param_name=\"InfraredLamp\";"  to set the camera back to auto. I checked the value in the db and it was set at 2. I rebooted the camera then renamed camera_setting.sh and set_configs.sh in the www/cgi-bin folder then copied your 2 new files into that folder. Reboot camera and tried to use the webui to change from auto to off. After saving the change to off a db query showed the value still set at 2. after a reboot the value was still 2. I am at a loss your db command in  camera_settings.sh is correct and I can manually change the value. It looks to me like a problem passing the value from the webui to the  camera_settings.sh script. Hope this helps.

gewinh, 2023-03-01:
Sorry to be using up your  time. I finally figured some of it out. It was to do with passing the parameter from the webui. I use ubuntu 22.04 as my desktop with brave as my default browser. All the problems I described were described using brave to access the webui, I haven't had problems before. When I switched to chrome to run the cameras webui, the update from auto to off runs fine and the query on the database shows a change from 2 to 1, do need a reboot to turn off the IR.. Don't understand why Brave would fiddle with passing parameters to a local web server, or maybe some sort of caching thing? So status is with your new files and a chrome as the browser the IR can get turned off and on. I tried on another camera with chrome  and it allowed me to change the setting in the database with the webui, but on reboot the IR was still physically on but the database showed 1 for the parameter. Same model camera same firmware. Tomorrow I will flash the sd again and try from scratch, as the camera that worked was the newly flashed sd from this morning.

roleoroleo, 2023-03-01:
It could be a cache problem.
Cam you try to force a clean of the cache?

gewinh, 2023-03-02:
The saga continues. I have not gone back to Brave and cleared the cache, but have been using chrome or firefox. These browsers(at the moment) let me change the IR led  field and save it and the change is correct when I query the database. The problem now is even though I have the IR led set to off and the database has a value of 1, the LED still is on regardless of reboot, and changing again(ie reset to auto, save, reboot, set to off, save, reboot. The IR light is always still on at night.  This is with fresh install of 0.1.5 on 2 separate cameras. I did notice something strange though on the fresh install of 0.1.5 on the sd card I didn't reset the camera but just restarted. The default value that came up for IR led on first boot was "off" not "auto", it must be retaining some info in the camera from the previous settings.  There is something definitely wrong in the passing of this parameter to ultimately set the camera operations. I will give it one last try tomorrow, doing a full reset on one of the cameras, a new 0.1.5 sd using chrome. Am I correct in assuming the changes you made to the 2 news scripts just related to timeout on the database updates, so if the parameters are being written correctly to the database then these changes will not fix the current problem. I have one camera out of 3 where I have successfully got the IR light off at night, same model and firmware as the others but I cannot reproduce this.

darkxst, 2023-03-02:
Sonoff-hack only updates the database, the rest is done by Sonoffs software.

roleoroleo, 2023-03-10:
I'm trying a different approach to change settings.

About mirror, instead of writing sql database directly, I'm trying to send an IPC message (UDP packet) to devctrl.
It seems working.

About IR it's possible to create these tmp files to control IR behavior:
```
[root@sonoff-hack]# touch /tmp/manualControl_autoIR
[root@sonoff-hack]# touch /tmp/manualControl_darkIR
[root@sonoff-hack]# touch /tmp/manualControl_brightIR
```

In both cases the change is in real time.

roleoroleo, 2023-03-11:
Check this commit:
https://github.com/roleoroleo/sonoff-hack/commit/a08e4c889da022ac35cae4b9660668d56e0be6a5

gewinh, 2023-03-12:
Looking great, I see what you are doing in the ipc section for the IR problem. I am building some new GK-200P2C cameras next week at my son's place  with HA, happy to field test 0.1.6beta if you can point me to an early release with these changes.

roleoroleo, 2023-03-12:
Here it is:
[GK-200MP2C_0.1.5.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/10951415/GK-200MP2C_0.1.5.tar.gz)

gewinh, 2023-03-20:
sorry haven't had much time. I did test the above copy of firmware on a GK-200MP2C running V0525.1.412build20220712, the IR controls worked without a reboot almost perfectly. The auto to on or off worked. The on to off  and back to on worked. Unfortunately you could not get back from either on or off to auto, when you did the save of a change from on to auto for example it would revert to on, never stay on auto. (hope that is clear). Also couldn't put my finger on the  reason, but did seem to have more issues with camera stability eg. camera going off line, disconnecting from mqtt and restarting. Not sure any of the changes use more memory or stress the cpu more. Playing with disabling ftp copy of action records and removing ftp . I now have a process running on a server nightly that uses scp to copy the days action records, then delete them from the camera. I am hoping just to give the camera less to do during the day to try and improve stability. The swap file option doesn't seem to help, I actually worry that the extra IO to the SD may be detrimental to the stability. Thanks again for the support.

roleoroleo, 2023-03-20:
Here it is:
[GK-200MP2C_0.1.5.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/11020676/GK-200MP2C_0.1.5.tar.gz)

gewinh, 2023-03-21:
Thanks I'll give it a try. While your improving things, A couple of other things I found about the release from last week, but didn't mention, was that the privacy switch works  for the camera.XXXXX_cam entity when displaying in HA but not for the webrtc stream rtsp://192.XXX.XXX.XXX:554 ...., this did work in the original 0.1.5 version. Also for the original 0.1.5 and the one from last week, the PTZ web interface is a pull your hair out challenge, if you want to use it to move the camera a reasonable distance and establish a setpoint to go to on boot. The camera freezes, reboots etc. Tried the soklutions mentioned previously without success. The yi_hack.ptz service in HA works a lot better. I know I am getting of topic on this particular issue, but not sure whether to open new issues for these things.

closed 2024-05-04 by github-actions[bot]