roleoroleo/sonoff-hack #138: S-CAM Documentation

couriersud, 2023-04-02:
I am running 0.1.5 on a fairly recent SONOFF firmware on the S-CAM model:

Firmware Version | 0.1.5
-- | --
Base Version | V5520.2053.0504build20230103
Model | S-CAM

In the Readme.md it is stated that the sd-card should be formatted using fat32 filesystem. But than in the ewelink app the sd-card is reported as unformatted.

What worked for this model was:

- remove all files from sd-card
- put sd-card in camera
- format using ewelink app
- continue with readme.md to install custom firmware

darkxst, 2023-04-02:
Formatting the sd-card in the ewelink app is certainly the fail proof way of doing it and people have reported having to do this previously. However I have always formatted sd-cards on my computer (Ubuntu) and never had a problem.

couriersud, 2023-04-03:
The format by the firmware creates an empty file `IsFormatted` in the root of the sd-card. As long as it does not exist, the eWelink app reports the sd-card as unformatted and blocks all sd-card related actions.

darkxst, 2023-04-04:
Sonoff-hack still boots without that file though? I assume it must since I dont have it on my sd-cards.

couriersud, 2023-04-04:
Yes, Sonoff-hack boots. But the sd-card is not recognized by the ewelink cloud. In the app, it is reported as unformatted and all sd-card related functionality won't work.

darkxst, 2023-04-04:
Does motion event recording work?

In anycase that would easy to check and fix in boot scripts
`[ ! -f /mnt/mmc/isFormatted ] && touch /mnt/mmc/isFormatted`

couriersud, 2023-04-11:
Yes, motion event recording strangely works. But eWeLink shows the SD card as unformatted.

2est, 2024-07-07:
But it looks like the motion sensor stops working. In my home assistant it doesn't respond in any way.
[image]

2est, 2024-07-08:
UPD: The problem with motion detection is solved by the "Motion Event Location" parameter (select 0 or 2)

kwai-chang, 2024-10-30:
> I answer myself. I confirm that the version 0.17 - GK-200MP2-B works correctly also for the S-CAM with the latest official firmware version as of today V5520.2053.0505 build20230320. My problem is that I didn't understand the instructions and I thought that I had to unzip the .tgz file and leave the .tar file in the root folder, but I also have to unzip the .tar file. I leave a screenshot of how it has to be the SD in case it helps someone in the future. The file "IsFormatted" that creates the APP when the SD is formatted must be left so that the camera recognizes the card correctly. Thank you very much for this great hack, thanks to this I will finally be able to use the camera in 100% offline mode. !

> where from do you get this:
The file "IsFormatted" that creates the APP when the SD is formatted must be left so that the camera recognizes the card correctly.

closed 2025-03-01 by github-actions[bot]