roleoroleo/sonoff-hack #144: Pre release 0.1.6 available

roleoroleo, 2023-04-20:
Please test the new pre-release:
https://github.com/roleoroleo/sonoff-hack/releases/tag/0.1.6

Below the changelog:

Bump devcontainer build to Debian 11 (thanks to @darkxst)
Bump gsoap to 2.8.104 and update mirrors (thanks to @darkxst)
Bump openssl to 1.1.1s (thanks to @darkxst)
Optimize gsoap build (thanks to @darkxst)
Optimise openssl build (thanks to @darkxst)
Remove broken Dockerfile and link it to the one used by devcontainers (thanks to @darkxst)
Fix scope for ONVIF Profile S (thanks to @darkxst)
Set Github actions to use a specific image (thanks to @darkxst)
Add missing mqtt-config
Add mqtt request to dump full configuration
Add timeout to sql insert and update
Fix onvif scope in service.sh
Complete rewrite of mqtt-sonoff (removed mqtt-config)
Add missing option for ir control
Change options from lower case to upper case
Move motion callback to a thread
Improve debug
Add ha discovery for switches and selects
Activate command line option for discovery
Add -a option to mqtt-sonoff if discovery is enabled
Enable rmdir in busybox
Add configurable ftp-port to ftppush.sh (thanks to @couriersud)
Enable rmdir in busybox (thanks to @couriersud)
Add inotifyd to busybox
Add new motion detection mode through inotify

roleoroleo, 2023-04-30:
Has anyone tried this version?
Do the changes made for motion detection work properly?

gewinh, 2023-05-01:
I did try this a few times,. I did get everything to work except the motion_detector sensor in HA setting motion location to 0 or 2 wasn't reflected in the UI after reboot as in the problem in 0.1.5. However the detector sensor status was unavailable in 0.1.6 whereas in 0.1.5 it worked OK. Tried a number of times  deleting everything fro HA rebooting and adding again still couldn't get the motion detector to work. I could get it to work in 0.1.5 with this firmware setting the motion location to 0.
Firmware Version	0.1.6
Base Version	V5520.2053.0505build20230320
Model	GK-200MP2-B

darkxst, 2023-05-01:
I'm testing on GK-200MP2-B. MQTT seems to be completely broken with location=2 for me.

My camera is also boot looping whenever I enable motion detection, need to investigate further, but I think it was doing this on 0.1.5 also.

gewinh, 2023-05-01:
I have found that with 0.1.5 on the GK-200MP2-B and GK-200MP2C I have to disable "snapshot for recorded video" in the webui before I can get stability. I have quite a few of these cameras running 0.1.5 with this disabled and integrated into HA. They seem solid, while just doing motion activated recording to the SD card. I use SCP to copy off/delete the previous days recording each night on these cameras and have turned off the ftp. Haven't had any unplanned reboots.
With 0.1.6 as stated above, with the same config. I cannot get the mqtt motion sensor to work.

roleoroleo, 2023-05-01:
I can't test the whole chain.
But if I set 2, when I create the file /tmp/colinkPushNotice with a touch, the event is fired properly.

darkxst, 2023-05-01:
OK, I was running beta of HA, there is some [bug parsing MQTT](https://github.com/home-assistant/core/issues/92322) messages currently. I downgraded back to stable release and its all working well with location 2.

@gewinh thanks that seems to have fixed the boot loops

darkxst, 2023-05-01:
There maybe an issue with motion sensitivity setting. This value is sent when turning off motion detection in the web UI.

```
Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:152
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: May 1, 2023 at 10:40:24 PM (2 occurrences)
Last logged: May 1, 2023 at 10:42:43 PM

Invalid option for select.sensitivity: 'OFF' (valid options: ['LOW', 'MEDIUM', 'HIGH'])
```

roleoroleo, 2023-05-04:
Yes, I know.
I will remove the mqtt feedback for this message.

darkxst, 2023-05-04:
> I will remove the mqtt feedback for this message.

or just update the HA Select to also support off (sensitivity 0), same should be done in the webui in that case. mqtt-sonoff is only thing that currently has the 'off' option.

roleoroleo, 2023-05-04:
You are right.
But I don't know what happens when the user selects Motion=on and Sensitivity=off.

closed 2024-05-03 by github-actions[bot]