roleoroleo/sonoff-hack #145: Is my (slow) web performances normal?

N0tMy4cc0unt, 2023-04-23:
I'm having two of this model, looks promising, but web responsiveness is going from 3 to over 30 seconds. By "ewb responsivenes" I'mean time for *.sh scripts to execute. It fill like something I'm missing as it should not be this slow (in average 8sec). PTZ responsiveness is OK, I see cameras are start moving shorty (less than 1 sec) after I initiate. Slowness of *.sh scripts is most noticeable on pages where I click on "Save" button at the bottom, often it take 8sec to complete save script and then similar time wait for page to acquire new parameters.
Is this behaviour normal or there is something I can do?
I already enabled swap, disabled motion detection and cloud, load is around 3...

darkxst, 2023-04-23:
The shell scripts themselves are quite slow, presumably due to slow file system (sd card) reads, which they do quite a lot of probing for all the different information.

roleoroleo, 2023-04-26:
Did you try another SD card?

darkxst, 2023-04-30:
> Did you try another SD card?

I seen this on both my cams, with a few different SD Cards.

The performance of the shell scripts is generally pretty slow. `df` is probably the slowest call, this can take 10secs sometimes. but even `get_config` calls can take 0.2-0.5 secs each.

closed 2024-05-03 by github-actions[bot]