roleoroleo/sonoff-hack #147: Preset Preview Lost

StarMonkey1, 2023-06-18:
When you create a preset the camera preview (live view) on that page becomes broken and the live view cannot be displayed. There is just a broken image icon.

Before creating the preset the live view on that page works fine.

roleoroleo, 2023-06-18:
Please, explain me better.

StarMonkey1, 2023-06-18:
[image]

StarMonkey1, 2023-06-18:
This page works fine before creating a preset but stops working after I've created a preset.

roleoroleo, 2023-06-18:
I tested it now and it seems working.
Which browser are you using?

StarMonkey1, 2023-06-18:
Latest version of Chrome. This happens with two Sonoff cameras, I've formatted the cards and done a fresh install and still have the same problem.

closed 2024-05-03 by github-actions[bot]