roleoroleo/sonoff-hack #151: wifi not connected

matew00, 2023-07-08:
Hello, I am running firmware version 0.1.5. Model GK-200MP2B and I cannot get wifi working. The wifi configuration is saved , but wifi strenght is 0%. Any suggestions please?

Nistp, 2023-08-07:
I'm having the same issue with 0.16. Everything works with an Ethernet cable, but the WiFi doesn't connect.

roleoroleo, 2023-08-19:
How did you configure wifi settings?
Through the app or from my webui?
Are you able to read the AP list from the webui? 
Opening this combo 
[image]
?

Nistp, 2023-08-21:
@roleoroleo I did it from the Web UI. I was able to read and select the AP list from there and when I entered the password and clicked on save it told me it saved. I later checked wpa_supplicant.conf and that wasn't updated - not sure if that's where the WiFi credentials get stored in.

matew00, 2023-08-22:
Hi roleo,

same here basically. version 0.1.5. Model GK-200MP2B

Opening combo/drop down list.
I pick SSID and PSK, save it. In webGUI, it remains saved.
When checking wpa_supplicant.conf, it does not get saved.

[root@sonoff-31dd]# cat /mnt/mtd/ipc/cfg/wpa_supplicant.conf
#20160318 ,New Interface From SmartLink Elian
ctrl_interface=/var/run/wpa_supplicant
update_config=1
network={
        ssid="0"
        psk="12345678"
}

matew00, 2023-08-22:
Hi Roleo,

update - tried to amend /mnt/mtd/ipc/cfg/wpa_supplicant.conf with the same ssid and psk a put into GUI. The result is still the same. No action to connect.
Output from ifconfig:

ra0       Link encap:Ethernet  HWaddr 30:4A:26:BC:31:DD
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:19 errors:0 dropped:4 overruns:0 frame:0
          TX packets:4 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2500 (2.4 KiB)  TX bytes:576 (576.0 B)

darkxst, 2023-08-23:
You can try run wpa_supplicant manually to find the problem
```wpa_supplicant -Dwext -d -ira0 -c/mnt/mtd/ipc/cfg/wpa_supplicant.conf```

You can probably also use `wpa_cli` to configure wifi (in case there is some error in your wpa_Supplicant.conf file)

darkxst, 2023-08-23:
Recreate wpa_supplicant.conf, with the following 2 lines
```
ctrl_interface=/var/run/wpa_supplicant
update_config=1
```
Kill wpa_supplicant if its running, then run
```
wpa_supplicant -Dwext -B -ira0 -c/mnt/mtd/ipc/cfg/wpa_supplicant.conf -P /var/pid_wpa_s
```
Then follow these instructions for `wpa_cli`
https://wiki.archlinux.org/title/wpa_supplicant#Connecting_with_wpa_cli

I've not tested this, but that should recreate a valid wpa_supplicant.conf for you.

roleoroleo, 2023-08-27:
When you save wifi options from webui, the settings are stored to the sqlite db.
You need a reboot to see them in the wpa_supplicant.

@darkxst Is it the same for your new model? Username and password are saved in the db?

darkxst, 2023-08-28:
They are set both in the db and wpa_supplicant.conf. I've never tried setup via son-hack though, so no idea if just the db setting is sufficient to get it working, but wpa_supplicant.conf is used at runtime to connect to wifi

matew00, 2023-09-06:
Hi Roleoroleo,

sure, I have always done a reboot after updating wifi settings from webgui, however it was never saved to wpa_supplicant.

roleoroleo, 2023-09-08:
Could you check the content of the db and the wpa_supplicant.conf after a reboot?
Does it match what you put in the webui?

blazejp83, 2024-04-03:
I started to have the exact same issues once I decided I want the camera to connect to a different WIFI. After long time, and different yi-hack versions, my GK-200MP2B was connected to a single WiFi, perhaps even connected from Yi app, can't remember. Yesterday, I wanted to change the WiFi. Camera couldn't connect.  I see constantly something like this:

[image]

I can change to a different Wi-Fi, enter password again, and again the same message (not connected). I've upgraded to 1.7 - same issue. Any ideas?

roleoroleo, 2024-04-03:
Are you connected with the ethernet cable?
When both wifi and eth are connected, the label shows "not connected".

closed 2024-12-13 by github-actions[bot]