roleoroleo/sonoff-hack #152: Discovered entities with a name that starts with the device name

DeadlySin2, 2023-08-03:
After the update to 2023.8.0 i get a warning: 

Some MQTT entities have an entity name that starts with the device name. This is not expected. To avoid a duplicate name the device name prefix is stripped of the entity name as a work-a-round. Please inform the maintainer of the software application that supplies the affected entities to fix this issue.

List of affected entities:

switch.livingroom_cam_motion_detection
select.livingroom_cam_sensitivity
select.livingroom_cam_ir
binary_sensor.livingroom_cam_motion
camera.livingroom_cam_image
switch.livingroom_cam_switch_on
switch.livingroom_cam_local_record
switch.livingroom_cam_rotate

Will it be fixed in later releases?

darkxst, 2023-08-03:
Its probably a bug, I don't think device name should in the entity names already.

can you "download diagnostics" from the mqtt device page for your camera and attach to this issue.

DeadlySin2, 2023-08-03:
[config_entry-mqtt-467bbd8aaf33487b44d85ab0925677cb.json.txt](https://github.com/roleoroleo/sonoff-hack/files/12247265/config_entry-mqtt-467bbd8aaf33487b44d85ab0925677cb.json.txt)
I have attached it to this post. It is not a bug - everything works fine. This is just a warning that it will causse a problem in version 2024.2.0

darkxst, 2023-08-03:
i meant a bug in the generated discovery messages.

As a workaround you can probably delete the HA Discovery "Name" prefix setting on the mqtt page in sonoff hack.

roleoroleo, 2023-08-19:
The workaround should work.
Probably the final fix is in mqtt-sonoff.c:

```
    if (mqtt_sonoff_conf.ha_name_prefix) {
        sprintf(dname, "%s %s", mqtt_sonoff_conf.ha_name_prefix, *suffix);
    } else {
        strcpy(dname, *suffix);
    }
```
Remove the 1st sprintf.

darkxst, 2023-08-22:
> Remove the 1st sprintf.

Yes, that looks the correct fix to me

closed 2024-05-03 by github-actions[bot]