roleoroleo/sonoff-hack #153: Updating stuff in the web interface doesn't necessarily update the system configurations

Nistp, 2023-08-07:
Hi, I'm using the sonoff-hack 0.1.6 (also tried with 0.1.5) with a GK-200MP2-B camera (FW: V5520.2053.0412build20220812) and I notice that a lot of the things I try to update thru the web-interface like the timezone or the RTSP port do not actually change. Are there any workarounds or fixes to that?

darkxst, 2023-08-07:
Its not possible to change the RTSP port on this camera.

Timezone handling changed in this camera, and not yet implemented in the web UI but there are workarounds, see the discussion in #107.

Nistp, 2023-08-07:
@darkxst thank you for your answer. 

Sounds good and yes, I was just looking at #107 and I've manually updated the timezone in config_timezone.ini file and it persisted even after a reboot.

closed 2023-08-07 by Nistp

darkxst, 2023-08-08:
I can confirm WiFi works when setup via eWeLink app, I havent tested directly setup via son-hack though. Its possible there were changes here with the newer V5520 firmware also.