roleoroleo/sonoff-hack #158: mqtt-sonoff crashes when colink.conf is not present

flreinhard, 2023-12-17:
mqtt-sonoff crashes when colink.conf is not present. (I fixed that, will submit pullrequest)

colink.conf is expected at /mnt/mtd/ipc/cfg/colink.conf

But that file does not exist. Maybe that file will be generated once the device was configured with the ewelink app?
Does anybody know?

Could somebody maybe provide a template for that file, so i could add code to generate the file if not present?

closed 2024-05-03 by github-actions[bot]

DeadlySin2, 2025-03-21:
Have the same issue
MQTT not working 

[root@Livingroom]# /mnt/mtd/ipc/app/s3Cloud --help
/mnt/mtd/ipc/cfg/colink.conf file not exist
curl version = 80294
4df076d5dec38a2288730b053c03b49af28d373fa48423e5da6ba9442ade1329
curl_easy_perform() failed: Couldn't resolve host name
/mnt/mtd/ipc/cfg/colink.conf file not exist
dd48e0517f183eb7c31f07aa88451a24e1da4ab35edb58e5b1fc86da15525a28
curl_easy_perform() failed: Couldn't resolve host name
/mnt/mtd/ipc/cfg/colink.conf file not exist
6d6c8087ead46e6377f8d37028b35b9d1054e2856fe5f400f21206b7a229a404
curl_easy_perform() failed: Couldn't resolve host name
/mnt/mtd/ipc/cfg/colink.conf file not exist
8d66216d445911f19df12faf46ce5490f50aaa038f9529b5390e505a41c91afc
curl_easy_perform() failed: Couldn't resolve host name
/mnt/mtd/ipc/cfg/colink.conf file not exist
df80ccd93c7e4236d65f9bc3bede2dcbb3c20fdaa6f38da89d78abb088e84430
curl_easy_perform() failed: Couldn't resolve host name

And Timezone change not working? i have tried several timezones by i only see +6 

[root@Livingroom]# cat /mnt/mtd/ipc/app/version
V5520.2053.0508build20230616

flreinhard, 2025-03-21:
got fixed with #185