roleoroleo/sonoff-hack #162: FW available doesn't appears in maintenance page

Dim3333, 2024-02-10:
Hello,

For an unknown reason, since some weeks (i don't known exactly), the firmware available doesn't appears  in maintenance page. So, the upgrade firmware is not possible.
The fimware installed is displayed.

[image]

josecarre, 2024-02-10:
Same happens to me, stuck on 0.1.5
[image]

roleoroleo, 2024-02-10:
Yes, it's a known problem.
This is why I upgraded busybox to 1.36.1
To upgrade the hack use the manual procedure:
https://github.com/roleoroleo/sonoff-hack/wiki/Manual-firmware-upgrade

Dim3333, 2024-02-11:
Thanks roleo.
It will be interesting for  a future release to integrate the possibility to upgrade firmawre (in maintenance page) by choosing a file and then update.

roleoroleo, 2024-02-11:
> I read the procedure but I don't inderstand where copy the file GK-200MP2B_x.x.x.tgz on the camera SDCard.

In the root folder of the card,
No subfolders.

Dim3333, 2024-02-11:
I done that before sent my last message and it doesn't works.

[image]

roleoroleo, 2024-02-11:
Ignore the string.
Click the button.

Dim3333, 2024-02-11:
I done.
[image]

roleoroleo, 2024-02-11:
If it doesn' work, backup your config and decompress the archive in your sd card (from your pc).
Just the sonoff-hack folder.
Then restore the config.

josecarre, 2024-02-11:
> Yes, it's a known problem. This is why I upgraded busybox to 1.36.1 To upgrade the hack use the manual procedure: https://github.com/roleoroleo/sonoff-hack/wiki/Manual-firmware-upgrade

Thanks, I can confirm this method worked in GK-200MP2B and GK-200MP2-B.

Dim3333, 2024-02-12:
> If it doesn' work, backup your config and decompress the archive in your sd card (from your pc). Just the sonoff-hack folder. Then restore the config.

It's works. Perfect.

closed 2024-03-03 by Dim3333