roleoroleo/sonoff-hack #163: preset call not working

dCypherNx, 2024-02-11:
I just updated to 0.1.7, but I can't call presets from either HA or ODM. In HA I can't even call pan tilt commands

roleoroleo, 2024-02-11:
It could be a problem with the ptz binary.
Open a ssh session and try to move the cam manually:
`/mnt/mmc/sonoff-hack/bin/ptz -a left -t 500`

If it doesn't work, try ptz_h and ptz_p biniaries.

dCypherNx, 2024-02-11:
the original ptz bin works through ssh but not for presets, I tried resetting them to the default hack and the same behavior persists, ODM is able to pan/tilt but not call or set presets, HA can't control any ptz aspect. When I try to set presets in ODM it returns "Too many presets"

DeadlySin2, 2024-02-12:
Can confirm the issue. On 0.1.6 ptz from HA works well. But on 0.1.7 i get error calling procedure error.

dCypherNx, 2024-02-12:
Seems that nothing changed

roleoroleo, 2024-02-12:
Too many preset is a strange message.
Please, post your /mnt/mmc/sonoff-hack/etc/ptz_presets.conf

dCypherNx, 2024-02-12:
there are few pretest, seems that onvif command is unable to access this file, at ODM the presets aren't lieted either

`0=0|969|956
1=1|2362|424
2=2|433|344
3=empty|-1|-1
4=empty|-1|-1
5=empty|-1|-1
6=empty|-1|-1
7=empty|-1|-1
8=empty|-1|-1
9=empty|-1|-1
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
- mnt/mmc/sonoff-hack/etc/ptz_presets.conf 1/10 10%`

roleoroleo, 2024-02-13:
Could you post the output of the list command?
`/mnt/mmc/sonoff-hack/bin/ptz -f /mnt/mmc/sonoff-hack/etc/ptz_presets.conf -a get_presets`

dCypherNx, 2024-02-13:
Just returned the usage guide, I tried ptz_h and it returned all presets, the ptz_p  " can't load library 'libptz.so'"
I renamed the ptz_h to ptz, it solved the problem at ODM, BUT at HA the behavior persists.
Can I control PTZ through mqtt?

roleoroleo, 2024-02-13:
Ok, now I understand.
Remove the file ptz and copy ptz_p as ptz.
It should fix the problem.

And I need to add a little fix to the boot script.

roleoroleo, 2024-02-13:
How did you add the cam to ha?
Are you using onvif standard or my integration?

dCypherNx, 2024-02-13:
Litle update about HA
Previously the presets are 0, 1, etc. Now, the presets are PresetToken_0, PresetToken_1, etc.
I'm using onvif standard, I didn't know you had your own integration LOL

roleoroleo, 2024-02-13:
Are presets working now?

This is my integration:
https://github.com/roleoroleo/yi-hack_ha_integration

dCypherNx, 2024-02-13:
Now the presets are working, but the single PTZ is not, just like in the h60ga, maybe the commands are also different, previously I was using UP, DOWN, LEFT, RIGHT of the HA, I will try some variations. In mean time I'll try your integration either

roleoroleo, 2024-02-13:
So pan and tilt in ODM work and in ha don't work. Correct?
Did you test all available pan and tilt in ODM?
I mean continuous, absolute and relative?

But...
I'm able to test it. I will let you know.

dCypherNx, 2024-02-13:
Presets are fine in both HA and ODM.I'll check variation of relative, absolute and continuous,

DeadlySin2, 2024-02-13:
I got ONVIF integration error after the update, so i removed ONVIF integration and tried to install default onvif integration and it can't connect to the camera - ONVIF: All connection attempts failed.

dCypherNx, 2024-02-13:
That's it! Previously we didn't need to set move_mode for single moves, now we need to.
Presets solved, single moves solved.

roleoroleo, 2024-02-13:
Another little fix:
[onvif_simple_server.gz](https://github.com/roleoroleo/sonoff-hack/files/14268357/onvif_simple_server.gz)

But I noticed a problem with ptz service in ha.
Ha doesn't ask to the cam the range of the pan and tilt movements.
When ha sends the command uses the range 0.0 - 1.0, but the cam works in degrees (0 - 360 and 0 - 180).
So the cam doesn't move properly.
I think ha is not compliant.

dCypherNx, 2024-02-19:
> Another little fix: [onvif_simple_server.gz](https://github.com/roleoroleo/sonoff-hack/files/14268357/onvif_simple_server.gz)
>
> But I noticed a problem with ptz service in ha. Ha doesn't ask to the [...]

They probably did it like this cause not all camera can pan 0-360 or tilt 0-180, so they implement as percentage. I don't know how onvif tells to implement

DeadlySin2, 2024-02-22:
I can't make ONVIF work on firmware 0.1.7. HA standart integration can't connect after update. YI-hack integration can't connect too. I roled back to 0.1.6 and HA standart integration connected, but YI-hack is not working.

roleoroleo, 2024-02-22:
Please, restart from scratch.
Clean your sd card and install again the hack.

closed 2024-06-22 by github-actions[bot]