roleoroleo/sonoff-hack #164: [Synology Surveillance Station] ONVIF don't work on 0.1.7

delvinru, 2024-02-22:
Hi, after upgrading from 0.1.6 to version 0.1.7, Synology cameras stopped working.

1. When connecting to the cameras, an error is issued about an incorrect username/password, although they are correct and specified in the camera settings:
[image]

2. When trying to connect to the camera, Synology returns error 464, which is quite atypical:
[image]

3. Well, it personally confuses me that ONVIF is defined on port 80, although in theory it should be at 1000:
[image]

roleoroleo, 2024-02-22:
Please, capture the traffic with tcpdump and send it to me.
Use a fake password.

delvinru, 2024-02-23:
Hi, I captured the traffic directly on the camera.

- `192.168.1.174` - camera
- `192.168.1.182` - synology
- `INVALID_USER_TEST:INVALID_PASSWORD_TEST` - test attempt for checking
- `root:securepassword` - valid pair

I hope this helps.

[capture_0.1.7.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/14386012/capture_0.1.7.tar.gz)

roleoroleo, 2024-02-23:
What do you mean exactly with "test attempt for checking" and "valid pair"?
The pair in the capture file is `INVALID_USER_TEST:INVALID_PASSWORD_TEST`.
I tested it and the digest is correct.
What's `root:securepassword`?

Another check.
Enable debug with
`echo 5 > /tmp/onvif_simple_server.debug`
and get the content of the files
`/tmp/onvif_simple_server.log`
after a client request.

delvinru, 2024-02-24:
Hi, maybe my mistake is that I didn't fully explain.

When I entered `INVALID_USER_TEST`, I specifically wanted to get a known error in the traffic. By "valid pair" I mean the correct username and password that are installed on the camera and should work in theory.

[onvif_simple_server.log.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/14392408/onvif_simple_server.log.tar.gz)

delvinru, 2024-02-24:
I also see from the logs that the server says that authentication was successful, but I also get an error from synology.
[image]

roleoroleo, 2024-02-24:
Ok, now it's clear.
This is what I understand from the capture.
- When you use INVALID_USER_TEST the cam replies with "not authorized" and it's ok.
- When you use root user the cam replies correctly, so the authentication is ok, confirmed by the logs.
- Then, a long sequence of commands: GetProfiles, GetCapabilities, GetServices, GetDeviceInformation, GetNetworkInterfaces, GetVideoSourceConfigurations, GetGuaranteedNumberOfVideoEncoderInstances, GetVideoEncoderConfigurations, GetVideoEncoderConfigurationOptions, GetVideoEncoderConfigurationOptions, GetProfiles, GetServiceCapabilities, GetProfiles.
- The problem starts when Syno requests CreateProfile method that it's not supported by the cam (no dynamic profiles allowed) and the cam replies with "The maximum number of supported profiles supported by the device has been reached"
- After that Syno asks again the same request but without authentication and the cam replies with "NotAuthorized".

So "incorrect username/password" it's a false message and the real problem is that CreateProfile is not supported.
I can change this behavior with a different response but I can't support profile creation.
So the real question is: "why the previuos version worked?"
This is the response of the previous daemon:
```
int MediaBindingService::CreateProfile(_trt__CreateProfile *trt__CreateProfile, _trt__CreateProfileResponse &trt__CreateProfileResponse)
{
    SOAP_EMPTY_HANDLER(trt__CreateProfile, "Media");
}
```
An empty message.

delvinru, 2024-03-03:
Sorry for the delayed answer. I tried to run the `onvif_simple_server`, but it crashes with an error (below logs)
I assume that this is due to the fact that patches were not applied during the build: https://github.com/roleoroleo/sonoff-hack/blob/master/src/onvif_simple_server/path.patch

```bash
1970-01-01 00:07:16 FATAL onvif_simple_server.c:282: Unable to find configuration file /etc/onvif_simple_server.conf
```

roleoroleo, 2024-03-09:
> In my opinion, you should not bother with dynamic profiles, this is not the level of cameras (IMHO)

I agree, but wich response should I send when Syno asks "CreateProfile"?
- Generic SOAP error
- "The maximum number of supported profiles supported by the device has been reached" SOAP error.
- Emtpy response

roleoroleo, 2024-03-09:
[action_failed.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/14545691/action_failed.tar.gz)
[maximum_number.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/14545693/maximum_number.tar.gz)
[empty_response.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/14545692/empty_response.tar.gz)
[no_response.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/14545694/no_response.tar.gz)

delvinru, 2024-03-09:
The archive contains traffic and small notes that I found during testing.

To put it briefly, `no_response.tar.gz ` it works as it should, login/password verification passes, the picture is transmitted without errors and the Synology interface does not issue errors.

[camera_test.tar.gz](https://github.com/roleoroleo/sonoff-hack/files/14545953/camera_test.tar.gz)

I also noticed that camera detection does not work (on all tested versions), although the ONVIF WS-Discovery option is enabled in the settings. It would be nice if we could sort out this problem as well.
[image]

roleoroleo, 2024-03-09:
The log confirms.
I think it's not a "standard" behavior, so I will add a special option to handle it.

About discovery, are you running in the same network?
It's a broadcast protocol, so it doesn't work if the nvr and the cam are in different subnets.
For example if you run the nvr inside a docker container (standard bridge network) it doesn't work.

delvinru, 2024-03-09:
> are you running in the same network?

Yes, ip of sonoff camera - `192.168.1.174`, synology - `192.168.1.182`, should be ok.

roleoroleo, 2024-03-09:
Does ODM detect the cam?

delvinru, 2024-03-09:
> Does ODM detect the cam?

no

roleoroleo, 2024-03-09:
Strange...
Is wsd_simple_server daemon running?
Did you select the correct interface?

delvinru, 2024-03-09:
I assume this is because of the following error (`/tmp/wsd_simple_server.log`):
```
1970-01-01 00:01:03 FATAL wsd_simple_server.c:453: Error joining multicast group
```

Why this error occurs, zero ideas.

roleoroleo, 2024-03-09:
Check the error number with this version:
[wsd_simple_server.gz](https://github.com/roleoroleo/sonoff-hack/files/14547269/wsd_simple_server.gz)

delvinru, 2024-03-09:
```
1970-01-01 00:01:00 FATAL wsd_simple_server.c:453: Error joining multicast group: 19
```

roleoroleo, 2024-03-11:
Are there other messages in the log?

delvinru, 2024-03-11:
No, just this line, and that's it.

roleoroleo, 2024-03-11:
Please, run the command with verbose log:
`/mnt/mmc/sonoff-hack/bin/wsd_simple_server --pid_file /var/run/wsd_simple_server.pid --if_name eth0 --xaddr http://%s/onvif/device_service -m sonoff_hack -n Sonoff -d 5`

closed 2024-07-11 by github-actions[bot]