roleoroleo/sonoff-hack #165: [GK-200MP2-B] No ONVIF

vlaraort, 2024-03-15:
Hi! thanks for the project!

I am trying to use the hack in a Sonoff GK-200MP2-B, updated with the last firmware, from a clean install.

All work fine, but the ONVIF functionality.

I have it set though the web interface to port 1000, and this configs

[image]

But when I check the open ports, the 1000 is not available, only

ftp 21
ssh 22
http 80
rtsp 554

The RTSP stream works fine, but Home Assistant cant find the camera with the ONVIF integration.

Any tips?

Camera model: GK-200MP2-B
Hack FW Version 0.1.7
Base version V5520.2053.0505build20230320
WiFi connection

roleoroleo, 2024-03-15:
With the last version, ONVIF shares the port 80 with the web server.

blazejp83, 2024-04-03:
I am having the same issue with ONVIF on 0.1.7. Tried to connect from my Synology, initally it doesn't detect anything, changed to port 80 as suggested, it connected but got message that there is no stream. Also if I use the function Test connection in my Synology NVR there's a message that connection to stream could not be established.

Also, if ONVIF is on port 80, then why in settings there's still configuration box with port 1000?

[image]

roleoroleo, 2024-04-06:
Synology requires some "specific" response from the cam.
See this post:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/843
I will share a new beta to test.

roleoroleo, 2024-04-06:
Here it is:
[GK-200MP2B_0.1.7.tgz](https://github.com/roleoroleo/sonoff-hack/files/14893232/GK-200MP2B_0.1.7.tgz)
[GK-200MP2-B_0.1.7.tgz](https://github.com/roleoroleo/sonoff-hack/files/14893233/GK-200MP2-B_0.1.7.tgz)
[GK-200MP2C_0.1.7.tgz](https://github.com/roleoroleo/sonoff-hack/files/14893235/GK-200MP2C_0.1.7.tgz)

RosensRauk, 2024-04-06:
Very nice hack but how do I turn off the snapshots/video-recordings activated from motion?

In camera settings I turned off snapshots by setting index: not record locally
My Sd -card gets full anyway.

I need the MQTT motion message feature for my HASSOS but want to deactivate saving snapshots/videos on my Sd-card.

roleoroleo, 2024-04-07:
Recording when a motion is detected is a feature handled by the original fw.
Take a look at the app, I don't remember if there is a specific switch for this.
Otherwise, add a line to crontab and delete all mp4 files.

closed 2024-08-07 by github-actions[bot]