roleoroleo/sonoff-hack #166: 0.1.7 scp doesn't seem to be working

gewinh, 2024-04-18:
Hi back after a long absence. I have had a number of Sonoff cameras running on 0.1.5 and integrated in to HA with great success for over 12 months( thanks for all the work)(this system has been frozen version wise for this period). Now in a moment of madness I have decided to update everything (pi5 ubuntu HA etc etc).  One of the changes has been to try 0.1.7 most things I have tested on a new Sonoff Cam slim using the GK-200MP2-B_0.1.7 are working well, except scp. ssh is working, if I ssh to a 0.1.5 camera and execute the scp command I get some stuff about usage, so the command is working. If I ssh to the 0.1.7 camera and execute the scp command I get "command not found".
Is this a path thing? 
I notice an issue raised for 0.1.6 to fix scp , is this the same issue for 0.1.7 if so how do I fix it.

gewinh, 2024-04-19:
Did some further investigation scp is in the 0.1.7 distribution. You can run it via /mnt/mmc/sonoff-hack/bin/scp.

Tried 2 temp fixes

A)Created a link to this by 
1) ssh to the camera 
2)cd /usr/bin 
3) ln -s /mnt/mmc/sonoff-hack/bin/scp
scp works
This doesn't survive a reboot

B)Looked at adding /mnt/mmc/sonoff-hack/bin to PATH
existing Path looks like
echo $PATH
/usr/bin:/usr/sbin:/bin:/sbin:/gm/bin:/gm/tools:/tmp/sd/yi-hack/bin:/tmp/sd/yi-hack/sbin:/tmp/sd/yi-hack/usr/bin:/tmp/sd/yi-hack/usr/sbin
PATH="$PATH:/mnt/mmc/sonoff-hack/bin"
echo $PATH
/usr/bin:/usr/sbin:/bin:/sbin:/gm/bin:/gm/tools:/tmp/sd/yi-hack/bin:/tmp/sd/yi-hack/sbin:/tmp/sd/yi-hack/usr/bin:/tmp/sd/yi-hack/usr/sbin:/mnt/mmc/sonoff-hack/bin
scp works
This change doesn't survive logout

Forgive my ignorance but I do not know how to make any of these changes permanent or how to update the distribution to include a fix. Any help?

gewinh, 2024-04-19:
A bit more info. 
I see /mnt/mmc/sonoff-hack/scripts/system.sh called by boot.sh
system.sh sets the path in
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/mnt/mmc/sonoff-hack/lib
export PATH=$PATH:/mnt/mmc/sonoff-hack/bin:/mnt/mmc/sonoff-hack/sbin:/mnt/mmc/sonoff-hack/usr/bin:/mnt/mmc/sonoff-hack/usr/sbin
but this path isn't there on boot, if it was run correctly and the path was set as above scp would work. I also noticed a problem with filezilla access giving sftp error. Again I would guess this is related to the path not being set correctly.

darkxst, 2024-04-19:
I think the path on ssh login is set in Dropbear

https://github.com/roleoroleo/sonoff-hack/blob/master/src/dropbear/localoptions.h

darkxst, 2024-04-19:
paths got messed up in this commit:
https://github.com/roleoroleo/sonoff-hack/commit/9bd2e3d5a475891a4857b25588e7a01ebc4b547e#diff-0c5450329c3e310a6a4ffd3041cfb1f656d34cfcc18a7052ea0b9d44f2acf371

gewinh, 2024-04-19:
Thanks. Yes I couldn't understand why all that yi-hack stuff  was in the path when i'm running the sonoff-hack. It's beyond my skills to fix that.  Hopefully someone will find the time to fix it.

darkxst, 2024-04-19:
@gewinh you could do a local build off PR #167 to test changes. I dont currently have access to my sonoff cams.

gewinh, 2024-04-19:
Thanks for doing this. Unfortunately I can get my head around how to do a local build. Install the toolchain? , vscode, development environment, dockers etc. etc. I think I will have to wait until the fix is incorporated into a release.

gewinh, 2024-04-20:
This old brain work a bit slowly these days. Finally sorted all the dependencies, paths etc. and got he updated 0.1.7 built. Installed on a GK-200MP2-B camera and scp and filezilla are now working. The PATH now looks correct.  Will test on the Sonoff 1080p slim cam and the HA integration later.  Thanks for all your help.

gewinh, 2024-04-20:
Tested on the Sonoff  1080p slim cam using firmware 0.1.7 all works fine, except for  camera on/off  as described in issue 0.1.7 camera on/off not working for some firmware

roleoroleo, 2024-04-20:
Sorry for this regression.
Below a binary with the fix posted by @darkxst:
[dropbearmulti.gz](https://github.com/roleoroleo/sonoff-hack/files/15048360/dropbearmulti.gz)

closed 2024-08-19 by github-actions[bot]

gewinh, 2025-03-19:
HI
Back again. Only seem to hassle you when I have a new camera. To date all my cameras are working well, thanks for the great work. Just installing a couple of sonoff mini cameras for my daughter, religiously went to your github site and download 0.1.7 set it up and away I went. The when I tried to use scp to backup recordings I came across the old Issue #166, the path issue error going to yi-hack folder when it is a sonoff camera. The brain then kicked into gear and I remember I had built a version of 0.1.7 to fix this last April as we discussed, so, I can fix the problem. Am I correct in assuming this path fix has not been applied to the official 0.1.7 download, thought this might catch out others, if it has dropped through the cracks. I know you must have a lot on your plate, thanks for all the great work.
Gavin

roleoroleo, 2025-03-19:
Sorry but the last release was from february.
But I should push a new release soon, because there are new prs made by other users that need to be released.

gewinh, 2025-10-31:
Testing my memory, it has been quite a while since I have setup a sonoff cam slim from scratch. I never had much success setting the wifi up using wifi_init.txt for the sonoff cameras. The way I setup my cameras was usually install them using ewelink on android. Get them on the network and get their Ip address, then reboot using the SDcard with the sonoff hack software. Http to the IP address and configure the camera. You can then either block ewelink or remove the camera from the the app. I also reserse the ip address in the router so I don't get problems later.

gewinh, 2025-10-31:
I got the wifi_init.txt to work on the Yi Home 2K Pro cameras but not on the sonoff's.