roleoroleo/sonoff-hack #168: 0.1.7 Camera on/off not working for some firmware

gewinh, 2024-04-20:
This issue is related to both the web interface and the HA integration camera on/off switches for Privacy
The 0.1.7 camera are running a process called "ProcessGuard" which restarts failed processes. As I understand the flow if you use the web interface or the HA integration to switch the camera off that runs the script privacy.sh which stops the rtsp streams and the alarmserver process. The camera stops recording video and there is no stream to view.  That works fine in 0.1.5. However in 0.1.7 privacy.sh is run the same way and the processes are stopped temporarily until the ProcessGuard restarts them in about 10 seconds. 
So although you have set the camera to privacy mode ie off, the camera is still operating.
I have made some changes to privacy.sh  for my cameras to overcome this issue.  Not necessarily the most efficient but it works. I have tested this script on camera running 0.1.7 and the camera now turn to privacy mode when you switch them off using the web or HA.
I tried to make the changes to the script backwardly compatible and I have tested it on 0.1.5 cameras and the privacy switch still works as intended.

Hopefully something like this can be included in a future release. Or alternatively a test for the status of the switch in the ProcessGuard process.

My new privacy script is

#!/bin/sh

RES=""

start_rtsp()
{
    /mnt/mtd/ipc/app/rtspd >/dev/null &
}

stop_rtsp()
{
    killall rtspd
}

start_alarmserver()
{
    ps | grep 'AlarmServer' | grep -v 'grep' | awk '{ printf $1 }' |xargs kill -CONT
}

stop_alarmserver()
{
    ps | grep 'AlarmServer' | grep -v 'grep' | awk '{ printf $1 }' |xargs kill -SIGSTOP
}

start_ProcessGuard()
{
    ps | grep 'ProcessGuard' | grep -v 'grep' | awk '{ printf $1 }' |xargs kill -CONT 2> /dev/null
}

stop_ProcessGuard()
{
    ps | grep 'ProcessGuard' | grep -v 'grep' | awk '{ printf $1 }' |xargs kill -SIGSTOP 2> /dev/null
}

if [ $# -ne 1 ]; then
    exit
fi

if [ "$1" == "on" ] || [ "$1" == "yes" ]; then
    if [ ! -f /tmp/privacy ]; then
        touch /tmp/privacy
        touch /tmp/snapshot.disabled
        stop_ProcessGuard
        stop_rtsp
        stop_alarmserver
    fi
elif [ "$1" == "off" ] || [ "$1" == "no" ]; then
    if [ -f /tmp/privacy ]; then
        rm -f /tmp/snapshot.disabled
        start_ProcessGuard
        start_alarmserver
        start_rtsp
        rm -f /tmp/privacy
    fi
elif [ "$1" == "status" ] ; then
    if [ -f /tmp/privacy ]; then
        RES="on"
    else
        RES="off"
    fi
fi

if [ ! -z "$RES" ]; then
    echo $RES
fi

roleoroleo, 2024-04-20:
If I check processes in my cam, ProcessGuard is not running.
And if I run it manually, it exits immediately:
```
[root@sonoff-hack]# /mnt/mtd/ipc/app/ProcessGuard
Create/Open Message Queue : 32768
ProcessGuard Exit by itself.
```
Strange...

But committed: https://github.com/roleoroleo/sonoff-hack/commit/5458f91a8b9d349c4bc29dfbc0e0d01df8404b87
Thank you for your code.

gewinh, 2024-04-21:
Yes it is strange, it took me a while to figure it out. Couldn't understand why I could turn off a camera on 0.1.5 but couldn't with 0.1.7.
See below re my testing on the same camera. Before upgrade to 0.1.7 the camera running 0.1.5 has no ProcessGuard after installing 0.1.7 the ProcessGuard app is running.
My logic is that the extra test in privacy.sh fixes the issue regardless of what version or camera you are running.
Thanks again for the great work and thanks for doing the commit

Firmware Version	0.1.5
Base Version	V5520.2053.0505build20230320
Model	GK-200MP2-B

no ProcessGuard 
ps -l | grep ProcessGuard
S     0 13462 12701  1456   336 pts0  10:26 00:00:00 grep ProcessGuard # just my ps grep query

/mnt/mtd/ipc/app/ProcessGuard
Create/Open Message Queue : 32768
ProcessGuard Exit by itself.

After upgrade to 0.1.7
Firmware Version	0.1.7

/mnt/mtd/ipc/app/ProcessGuard
Create/Open Message Queue : 0

yes ProcessGuard is now running

ps -l | grep ProcessGuard
$     0   221     1   588   196 0:0   10:20 00:00:00 /mnt/mtd/ipc/app/ProcessGuard  #processguard is now running
S     0  2514  2300  1468   352 pts0  10:25 00:00:00 grep ProcessGuard #just my query

gewinh, 2024-04-25:
Another small issue for a rainy day. If you turn off the camera from either HA or the web interface. eg. you are home and want the inside cameras off. All works well privacy.sh has run and the rtsp streams and recording are off. However, if  the camera restarts for some reason eg power interruption, the camera starts up with the default setting of on. ie rtsp streams  and recording are now on. I assume the state of camera switches are stored somewhere, is it possible to read the old state of the on/off switch on reboot so the camera comes backup with the setting it had before the restart? I can certainly live with the way it is, cameras don't get restarted often, but some people might get caught out thinking the camera is off when its not. I run everything from HA, when I turn on the input _boolean home, I turn off/on a lot of things, inside cameras are turned off. I assume they stay off while this boolean is still on. There are HA solutions like re running the a HA Script if a camera state changes, but would prefer to fix it at the source.

closed 2024-08-24 by github-actions[bot]