roleoroleo/sonoff-hack #169: Feature Request: Add ONVIF MoveStatus support for use with frigate autotracking

minego, 2024-05-02:
After finding this project I bought a GK-200MP2-B in hopes of getting it playing nicely with Frigate (and Home Assistant.) I have it working very nicely so far, but I have been unable to get auto tracking to work.

When I try, I see this in the frigate logs:

```
frigate.ptz.autotrack          WARNING : Disabling autotracking for kitchen: ONVIF MoveStatus not supported
```

I'm hoping that this can be added without too much difficulty. If no one has time to work on this, then please point me in the right direction and I'll see if I can make a PR (when I can find time...)

roleoroleo, 2024-05-07:
This problem was already fixed.
But not yet released.

roleoroleo, 2024-05-08:
Tested today with my cam, it works.
I will release a new version asap.

roleoroleo, 2024-05-10:
Try this beta:
[GK-200MP2B_0.1.7.tgz](https://github.com/roleoroleo/sonoff-hack/files/15272617/GK-200MP2B_0.1.7.tgz)
[GK-200MP2-B_0.1.7.tgz](https://github.com/roleoroleo/sonoff-hack/files/15272619/GK-200MP2-B_0.1.7.tgz)
[GK-200MP2C_0.1.7.tgz](https://github.com/roleoroleo/sonoff-hack/files/15272620/GK-200MP2C_0.1.7.tgz)

klipper-vp, 2024-08-15:
*here unfortunately it didn't work giving this error*

Timestamp
2024-08-15 16:21:29
Tag
frigate.ptz.autotrack
Message
Disabling autotracking for sonoff-camera: ONVIF MoveStatus not supported

Firmware Version | 0.1.7
Base Version | V2524.1.412build20220712
Model | GK-200MP2B

closed 2024-12-14 by github-actions[bot]