roleoroleo/sonoff-hack #173: How to restart the camera via HTTP request?

SirSheik, 2024-06-20:
Hello.
Sorry, I don't know if this has already been discussed, but I was looking in previously opened/closed Issues but I didn't see it.
Is it possible to reboot the camera via HTTP request? I would like to automate from Node-Red the restart of the camera if it stops responding, because the camera crashes from time to time, I guess because it is constantly sending video over RTSP.
Thank you very much
Best regards

roleoroleo, 2024-06-29:
Check the wiki:
https://github.com/roleoroleo/sonoff-hack/wiki/Web-services-description#maintenance

SirSheik, 2024-07-01:
Hello.
Sorry I didn't know about the existence of that wiki, I thought that all the information available was only Readme. My mistake.
If you accept a suggestion, maybe it would be good to put a link in the readme to the wiki.
Thank you very much for your help.

closed 2024-07-01 by SirSheik