roleoroleo/sonoff-hack #176: Connection refused (http and ssh) after setting authentication

msaitz, 2024-10-02:
Hello! I'm using S-CAM with the hack version `0.16`.  Yesterday I added auth using the webui fields noted below and restarted the camera. What I found is that I'm able to view the rtsp stream using the new credentials, but the change somehow made the port 80 and 22 to no longer work (see screenshot)

What could've happened? any idea of how can I re-gain access to the camera without resorting on a reset?

I have another unused camera for trying to reproduce it, both with `0.16` and `0.17`.
 
[image]

[image]

Thanks for your work!

roleoroleo, 2024-10-04:
Remove the sd card and check the file /sonoff-hack/etc/system.conf
Are your settings properly saved?

msaitz, 2024-10-05:
Interestingly that file had the settings messed up with ssh/http disabled, and the cloud re-enabled plus some other changes. 
It maintained the root password and the newly added user/pass.

Editing that back to the expected content has fixed it thanks. No idea of what may have happened!

closed 2024-10-05 by msaitz