roleoroleo/sonoff-hack #180: Configuring GK-200MP2-B (50520230320) without Wifi

jpr105, 2024-12-31:
Hello,

I'm new in Home Assistant and I was looking for an IP camera to use locally on my network.

I bought the Sonoff GK-200MP2-B because I found its functionality interesting for a reasonable price.

I also found all the features of your hack very useful, thanks 😉

Unfortunately I can't install it. Here's what I did:
1 - Connecting to eWeLink: a) wifi settings not entered in eWelink app because I don't trust this app b) add the camera using the QR code and my RJ45 cable c) each time I reboot, the camera asks me to configure the wifi but it finally works (LAN).

2 - Camera information :
Version: 50520230320
Model: GK-200MP2-B
FW Name: GK-200MP2D

3 - To install your hack I chose GK-200MP2-B_0.1.7.tgz here https://github.com/roleoroleo/sonoff-hack/releases

4 - I flashed the hack on a 4 GB SD card on Linux following all the instructions

Unfortunately it doesn't work.

Maybe GK-200MP2C which is close to GK-200MP2D ???

Thanks for your job

Regards - Jean-Paul

roleoroleo, 2025-01-01:
I can't test the hack with this cam.
I don't own it.
Are you able to make a backup? I can give a look.

jpr105, 2025-01-01:
OK, I can make a backup, but from what and how?
Can you explain me your needs?

Doubledom45, 2025-01-04:
J'écris directement en Français !
As-tu décompilé le fichier correctement, des fois le fichier décompilé  `GK-200MP2-B_0.1.7` n'as pas l'extension `tar`
Le transformer en `GK-200MP2-B_0.1.7.tar` puis le décompilé !
Il faut avoir 
[image]

jpr105, 2025-01-04:
Oui j'ai bien la structure correspondante à l'image.

De toute façon j'ai fait un retour Amazon :

- obligé de mettre en place un hack pour débloquer les fonctionnalités intéressantes
- utiliser une app (eWeLink) même pas propriété Sonoff avec de la pub dedans et des options payantes

Ça commence à faire beaucoup 😳

Du coup je me tourne vers Amcrest : https://www.amazon.com/Amcrest-UltraHD-Security-4-Megapixel-IP4M-1041W/dp/B095XCGKTF?th=1
Pas dispo en France mais livrable via Amazon USA

Les commentaires ont l'air dithyrambiques et les intégrations Home Assistant et Synology directement "out of the box"

@+

-------------------

Yes, I got the structure corresponding to your image.

Anyway, I've returned it to Amazon :

- you have to set up a hack to unlock interesting features
- use an app (eWeLink) that isn't even Sonoff property with ads in it and paid options

That's getting to be a lot 😳

So I'm turning to Amcrest: https://www.amazon.com/Amcrest-UltraHD-Security-4-Megapixel-IP4M-1041W/dp/B095XCGKTF?th=1
Not available in France but available via Amazon USA

The comments seem enthusiastic and Home Assistant and Synology integrations are straight out of the box.

Ciao

Doubledom45, 2025-01-04:
OK dommage !
Faire l'issue si possible ! ou @roleoroleo

roleoroleo, 2025-01-04:
> OK, I can make a backup, but from what and how? Can you explain me your needs?

If you have a shell access, run the dd command for each available partition.
For example:

```
root@cam:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00060000 00001000 "uboot"
mtd1: 001d0000 00001000 "boot"
mtd2: 00120000 00001000 "rootfs"
mtd3: 00340000 00001000 "home"
mtd4: 00130000 00001000 "backup"
mtd5: 00020000 00001000 "env"
mtd6: 00010000 00001000 "mfg"
mtd7: 00010000 00001000 "conf"

dd if=/dev/mtdblock0 of=/mnt/mmc/mtdblock0.bin
dd if=/dev/mtdblock1 of=/mnt/mmc/mtdblock1.bin
...
dd if=/dev/mtdblock7 of=/mnt/mmc/mtdblock7.bin
```

And share with me the files (6 files if I remember correctly).

jpr105, 2025-01-04:
Sorry, but the camera is already packed to be returned to Amazon by post.

roleoroleo, 2025-01-04:
No problem.

jpr105, 2025-01-04:
Thanks, so I close the issue

closed 2025-01-04 by jpr105