roleoroleo/sonoff-hack #181: Can't disable local record

sonped, 2025-01-03:
I can't disable local recording in version 0.17.

[image]

https://github.com/user-attachments/assets/d5cc8df1-d057-47f2-8938-2c50e546a54a

flreinhard, 2025-03-09:
Hello @sonped !

I opened an issue for LocalRecord but just the other way round #193. I was not able to disable it.
Background:
- The gui is populated from a value read from the database
- on your firmware the ipc command that will alter the database value will probably fail due to a missing binary `AVRecSch`
- in consequence, no matter what you click, you can't disable local record

Does your camera do local recordings?

Maybe you can provide some information on the directory listing also in issue #193 thx!

sonped, 2025-03-09:
@flreinhard 
Yes camera does local recording which I don't need.

Not sure which directory you what me to provide here, so here is what I can see might be off interest based on #193

[root@sonoff-fee8]# netstat -anp | grep 19000
netstat: /proc/net/tcp6: No such file or directory
netstat: /proc/net/udp6: No such file or directory
netstat: /proc/net/raw6: No such file or directory

[root@sonoff-fee8]# ls -la /mnt/mtd/ipc/app/
drwxrwxrwx   11 root     root           389 Mar 20  2023 .
drwxr-xr-x    3 root     root            51 Jul 29  2021 ..
-rwxr-x---    1 root     root        613668 Mar 20  2023 AlarmServer
-rwxr-x---    1 root     root        228076 Mar 20  2023 IOTCare
-rwxrwx---    1 root     999           5864 Mar 20  2023 ProcessGuard
-rwxr-x---    1 root     root        636168 Mar 20  2023 avencode
-rwxr-x---    1 root     root          7328 Mar 20  2023 captive_server
drwxrwxrwx    3 root     root           259 Mar 20  2023 cfg
-rwxrwx---    1 root     999          26376 Mar 20  2023 checkMid
-rw-r--r--    1 root     root             0 Jan  1  1970 colink
-rwxrwx---    1 root     999           2875 Aug  1  2022 colink.sh
-rwxrwx---    1 root     999           1630 Aug  1  2022 colink_register.sh
-rwxrwx---    1 root     999            530 Aug  1  2022 colinkwtg.sh
drwxrwxrwx    2 root     root            58 Mar 20  2023 dbback
-rwxrwx---    1 root     999         156508 Mar 20  2023 devctrl
drwxrwxrwx    2 root     root           271 Dec 11  2020 drive
-rwxr-x---    1 root     root        133760 Mar 20  2023 factorytest
drwxrwxrwx    2 root     root            93 Dec 11  2020 font
drwxrwxrwx    2 root     root            38 Dec 15  2020 lib
drwxr-xr-x    3 root     root            27 Aug 27  2021 reserved
-rwxr-x---    1 root     root        137576 Mar 20  2023 rtspd
-rwxr-x---    1 root     root       2032260 Mar 20  2023 s3Cloud
drwxrwxrwx    2 root     root           306 Mar  2  2022 script
drwxr-xr-x    2 root     root           125 Dec 11  2020 sd
drwxr-xr-x    4 root     root            96 Feb 26  2022 snd
-rw-r--r--    1 root     root            29 Mar 20  2023 version

flreinhard, 2025-03-09:
could you please add your camera model and firmware version?
cat /mnt/mtd/ipc/app/version

sonped, 2025-03-09:
I can tomorrow, but it's already in the first post.

ZeynepKlc, 2025-06-26:
After connecting to the camera via SSH, run the following commands one by one.
Make sure to enter each command in sequence, and reboot the device at the end.

- alias sqlite3='/mnt/mmc/sonoff-hack/bin/sqlite3'

- sqlite3 /mnt/mtd/db/ipcsys.db "SELECT * FROM t_record_plan;"

- sqlite3 /mnt/mtd/db/ipcsys.db "UPDATE t_record_plan SET c_enabled=1 WHERE c_store_path='/mnt/mmc';"

- reboot

After the reboot, you should see that local recording has been enabled.

If you want to disable local recording again, run the following command and then reboot the device:

- sqlite3 /mnt/mtd/db/ipcsys.db "UPDATE t_record_plan SET c_enabled=0 WHERE c_store_path='/mnt/mmc';"
- reboot

closed 2025-11-20 by github-actions[bot]