roleoroleo/sonoff-hack #182: Wi-Fi configuration on GK-200MP2-B

f870103, 2025-01-04:
## Description
After configuring Wi-Fi on the Sonoff GK-200MP2-B camera, the device appears to connect to the Wi-Fi Access Point (AP), but it does not obtain an IP address via DHCP. As a result, it is inaccessible over the Wi-Fi network.

## Steps to Reproduce
1. Directly install the hack, **skipping the setup in the official app**.
2. Connect the camera to the network via Ethernet.
3. Access the camera’s WebUI through its assigned Ethernet IP address.
4. Enter the Wi-Fi credentials in the WebUI and save the configuration.
5. Disconnect the Ethernet cable.

## Expected Behavior
The camera should connect to the Wi-Fi AP and obtain an IP address via DHCP, making it accessible over the Wi-Fi network.

## Actual Behavior
The camera connects to the Wi-Fi AP (confirmed by the router), but it does not receive an IP address via DHCP.
[image]

The camera remains inaccessible over the Wi-Fi network.

## Environment
Camera Model: Sonoff GK-200MP2-B
Hack Version: https://github.com/roleoroleo/sonoff-hack/commit/041803a4351b4313d16235c0db381b904d94ff6b
Network Configuration:
* Security: WPA2 PSK
* DHCP: Enabled

## Additional Information
The camera is visible on the router's connected devices list, but no IP is assigned.

I have tested on 3 different cameras and all have the same result.

roleoroleo, 2025-01-04:
And what about, if you reboot the cam without the ethernet cable at the end of the procedure?

f870103, 2025-01-04:
> And what about, if you reboot the cam without the ethernet cable at the end of the procedure?

Same result, connected to router but no IP assigned.

roleoroleo, 2025-01-04:
I have to check the code.

flreinhard, 2025-03-01:
This issue is resolved with pullrequest #179 
See: https://github.com/roleoroleo/sonoff-hack/blob/f83fcd8d38a2151c376e2a9485183fbdf07d8fa7/sdhack/boot.sh#L60

There's no current release yet, but you could try an artifact from a github build:
https://github.com/roleoroleo/sonoff-hack/actions/runs/13608127300/job/38042011935
there's a link to download the artifacts of that build:
https://github.com/roleoroleo/sonoff-hack/actions/runs/13608127300/artifacts/2675855101

roleoroleo, 2025-03-14:
I will publish a new release the next weekend.

closed 2025-07-13 by github-actions[bot]