roleoroleo/sonoff-hack #183: Unable to download firmware file.

multiflexi, 2025-01-07:
Exactly what title says happens when attempting to do online update. I have version 1.5 on 3 GK-200MP2-B cameras. How can update it?

roleoroleo, 2025-01-12:
This log is here:
https://github.com/roleoroleo/sonoff-hack/blob/041803a4351b4313d16235c0db381b904d94ff6b/src/www/httpd/cgi-bin/fw_upgrade.sh#L85
wget command fails.
Is there ehogh space in the sd card?
Is the cam able to reach github.com?

MichielioZ, 2025-01-20:
I think maybe something is wrong with checking for an updated version.
Not sure if it's the same issue that @multiflexi had, but I get:

[image]

multiflexi, 2025-01-20:
That is exactly what I see. I have plenty of space on the SD card and the network works fine, I can resolve addresses and ping google.com etc.

[image]

MichielioZ, 2025-01-20:
> That is exactly what I see.

OK, so it's the same problem I have. Your original comment was confusing to me, as it suggested you actually had an error while upgrading, when instead you are trying to upgrade to version "" (=empty string), which obviously doesn't work.

roleoroleo, 2025-01-21:
Probably I remember now.
There was a issue in wget fixed with busybox 1.36.1.
You need to update the cam manually.
Backup your configuration and install the hack from scratch.

closed 2025-05-22 by github-actions[bot]