roleoroleo/sonoff-hack #184: "Go to Presets" error

renegade304, 2025-01-25:
The "Go to Presets" command does not work on positions 0 - 9, even though they are saved. Preset numbers 10 - 14 work correctly.

roleoroleo, 2025-01-26:
10-14 are predefined presets.
Check if your presets are properly saved:
`cat /mnt/mmc/sonoff-hack/etc/ptz_presets.conf`

renegade304, 2025-01-27:
The file content looks like this:

0=Start|2364|-102
1=OK|2329|-182
2=Domka|2632|-182
3=Nelka|2053|-412
4=empty|-1|-1
5=empty|-1|-1
6=empty|-1|-1
7=empty|-1|-1
8=empty|-1|-1
9=empty|-1|-1

roleoroleo, 2025-01-28:
Mmmmh
Negative numbers are not allowed.
ptz binary discards them.
Please try to set presets again.

renegade304, 2025-01-28:
I tried - it was saved as a negative number again.
When I manually deleted the minus - the preset location setting worked.
It looks like an error while saving the preset.

roleoroleo, 2025-02-01:
When the camera starts up, does it do a full pan and tilt?

flreinhard, 2025-03-01:
I think, there's a bug in ptz.c https://github.com/roleoroleo/sonoff-hack/blob/f83fcd8d38a2151c376e2a9485183fbdf07d8fa7/src/ptz/ptz/ptz.c#L572-L577
I think it should be
```C
        x = preset_buffer[3];
        y = preset_buffer[4];
        if (x < MIN_X) x = MIN_X;
        if (x > MAX_X) x = MAX_X;
        if (y < MIN_Y) y = MIN_Y;
        if (y < MAX_Y) y = MAX_Y;
```
otherwise your limits to min(0) and max won't work.

@roleoroleo: why do you limit to 0 when /dev/ptz delivers negative numbers?

flreinhard, 2025-03-02:
@renegade304 you give my changes a try. you can download a file at the bottom here: https://github.com/roleoroleo/sonoff-hack/actions/runs/13616982538

roleoroleo, 2025-03-06:
> I think, there's a bug in ptz.c 

I don't rememer why I added this check but you are right: it's wrong.
Or it's useless...
I'll fix it.

> @roleoroleo: why do you limit to 0 when /dev/ptz delivers negative numbers?

Because ptz can't be negative, at least my model (GK-200MP2B with firmware version V2524.1.245build20191030).
0 <= x <= 3500
0 <= y <= 900

When is negative is because you asked to the cam to go beyond the limit.
For example: if you run a left command and the cam reaches the limit, the cam can't move but the posizion counter continues to decrease until you stop it. After this step, all data position all wrong.

PR #189 (2025-03-06): #184 Avoid negative numbers in PTZ presets
Fixes #184
closed 2025-03-06 by roleoroleo