roleoroleo/sonoff-hack #193: Local Record not working

flreinhard, 2025-03-05:
disclaimer: the camera has not been configured with ewelink, goal of the issue is to gather information what's missing on a non-configured camera to make local-record work there too.

Several issues:
- from page=camera_settings it's not possible to sabe "LOCAL_RECORD=yes", background:
  - `camera_settings.sh` calls `ipc_cmd -l on`
  - ipc_cmd sends some (undocumented) magic udp packet to some service listening on port 19000
  - there's nothing listening on port 19000 on my camera
  - there's no update in ipcsys.db
- which process is not runnig on my camera?
  - there are only a few binaries in /mnt/mtd/ipc/ that link to libdbmanager.so that could do that:
    - AlarmServer
    - IOTCare
    - avencode
    - devctrl
    - factorytest
    - rtspd
    - s3Cloud
    - sd/sd_detect
  - not running on my camera:
    - IOTCare
    - devctrl
    - factoryreset
    - s3Cloud
  - conclusion
      - IOTCare => no, only wierd port numbers
      - devctrl => no, port 14000, but i would not be able to rotate
      - s3Cloud => no, no even port numbers

The sonoff-hack sources, port 19000 should belong to "AVRECSCH", but i can't find that binary on my camera.

So finally, i've no idea, which service is missing => HELP PLEASE :)

roleoroleo, 2025-03-07:
This is my output:
```
[root@sonoff-hack]# netstat -anp | grep 19000
netstat: /proc/net/tcp6: No such file or directory
udp        0      0 127.0.0.1:19000         0.0.0.0:*                           314/AVRecSch
netstat: /proc/net/udp6: No such file or directory
netstat: /proc/net/raw6: No such file or directory
```
and
```
[root@sonoff-hack]# ls -la /mnt/mtd/ipc/app/
drwxrwxrwx   10 root     root           398 Jul 12  2022 .
drwxr-xr-x    5 root     root             0 Oct 17  2019 ..
-rwxr-xr-x    1 root     root         22468 Jul 12  2022 AVRecSch
-rwxr-xr-x    1 root     root        552560 Jul 12  2022 AVRecorder
-rwxr-xr-x    1 root     root         88460 Jul 12  2022 AlarmServer
-rwxrwxrwx    1 root     root         10419 Jul 12  2022 App.sh
-rwxr-xr-x    1 root     root        209708 Jul 12  2022 IOTCare
-rwxrwxrwx    1 root     root          5864 Jul 12  2022 ProcessGuard
-rwxrwxrwx    1 root     root         28229 Jun  6  2020 av.xml
-rwxrwxrwx    1 root     root        760620 Jul 12  2022 avencode
-rwxr-xr-x    1 root     root          7308 Jul 12  2022 captive_server
drwxrwxrwx    2 root     root           188 Jul 12  2022 cfg
-rw-r--r--    1 root     root             0 Jan  1  1970 colink
-rwxr--r--    1 root     root          2319 Jun  8  2022 colink.sh
-rwxr--r--    1 root     root           531 Jul 17  2021 colinkwtg.sh
drwxrwxrwx    2 root     root            32 Jul 12  2022 dbback
-rwxrwxrwx    1 root     root        169840 Jul 12  2022 devctrl
drwxrwxrwx    2 root     root           301 Jul 12  2022 drive
-rwxrwxrwx    1 root     root         96588 Jul 12  2022 factorytest
drwxrwxrwx    2 root     root            93 Jul 12  2022 font
-rwxrwxrwx    1 root     root          7256 Jul 12  2022 led_crontabs
drwxrwxrwx    2 root     root           435 Jul 12  2022 lib
drwxrwxrwx    4 root     root            43 Jul 12  2022 res
-rwxrwxrwx    1 root     root        616960 Jul 12  2022 rtspd
drwxrwxrwx    2 root     root           140 Jul 12  2022 script
drwxr-xr-x    2 root     root           125 Jul 12  2022 sd
-rwxrwxrwx    1 root     root            25 Jul 12  2022 version
```

flreinhard, 2025-03-09:
Here's what i got:

```
[root@TestCam]# cat /mnt/mtd/ipc/app/version 
V5520.2053.0412build20220905
```

```
[root@TestCam]# ls -la /mnt/mtd/ipc/app/
drwxrwxrwx   11 root     root           389 Sep  5  2022 .
drwxr-xr-x    3 root     root            51 Jul 29  2021 ..
-rwxr-x---    1 root     root        613668 Sep  5  2022 AlarmServer
-rwxr-x---    1 root     root        223964 Sep  5  2022 IOTCare
-rwxrwx---    1 root     999           5864 Sep  5  2022 ProcessGuard
-rwxr-x---    1 root     root        636164 Sep  5  2022 avencode
-rwxr-x---    1 root     root          7328 Sep  5  2022 captive_server
drwxrwxrwx    3 root     root           259 Sep  5  2022 cfg
-rwxrwx---    1 root     999          26376 Sep  5  2022 checkMid
-rw-r--r--    1 root     root             0 Jan  1  1970 colink
-rwxrwx---    1 root     999           2875 Aug  1  2022 colink.sh
-rwxrwx---    1 root     999           1630 Aug  1  2022 colink_register.sh
-rwxrwx---    1 root     999            530 Aug  1  2022 colinkwtg.sh
drwxrwxrwx    2 root     root            58 Sep  5  2022 dbback
-rwxrwx---    1 root     999         156416 Sep  5  2022 devctrl
drwxrwxrwx    2 root     root           271 Dec 11  2020 drive
-rwxr-x---    1 root     root        133760 Sep  5  2022 factorytest
drwxrwxrwx    2 root     root            93 Dec 11  2020 font
drwxrwxrwx    2 root     root            38 Dec 15  2020 lib
drwxr-xr-x    3 root     root            27 Aug 27  2021 reserved
-rwxr-x---    1 root     root        137576 Sep  5  2022 rtspd
-rwxr-x---    1 root     root       2032252 Sep  5  2022 s3Cloud
drwxrwxrwx    2 root     root           306 Mar  2  2022 script
drwxr-xr-x    2 root     root           125 Dec 11  2020 sd
drwxr-xr-x    4 root     root            96 Feb 26  2022 snd
-rw-r--r--    1 root     root            29 Sep  5  2022 version
```
Major difference i see: `AVRecSch` and `AVRecorder` are missing, but there's `s3Cloud`. I started that binary, that's the commandline output:
```
[root@TestCam]# /mnt/mtd/ipc/app/s3Cloud --help
09:07:51 [main.cpp checkProcessIsRunning:283] E> myserver is not running! begin to run..... pid=12699
09:07:51 [main.cpp main:311] E> 000 enter into s3cloud main 
09:07:51 [s3Cloud.cpp InitS3AvStream:825] E> Notice the response port:55138  g_nObsVideoPort:55138
09:07:51 [serviceinterface.cpp Call_Service:140] D> ModuleId:1,communicationPort:11000,cmdId:1001
09:07:51 [s3Cloud.cpp ThreadAVS3StreamRecv:370] E> Real-time audio and video streaming callback service for S3 
09:07:51 [circleframe.cpp InitMem:76] E> InitMem 200 
09:07:51 [circleframe.cpp InitMem:93] D>  nShmKey = 336265546 
09:07:51 [circleframe.cpp InitMem:136] E> m_nVideoShareMemId =65538,0x764e8000,20
09:07:51 [circleframe.cpp InitMem:155] D> m_ShareAVDataLen:345600,345568 
09:07:51 [circleframe.cpp InitMem:76] E> InitMem 500 
09:07:51 [circleframe.cpp InitMem:93] D>  nShmKey = 839582026 
09:07:51 [circleframe.cpp InitMem:136] E> m_nVideoShareMemId =0,0x764d4000,50
09:07:51 [circleframe.cpp InitMem:155] D> m_ShareAVDataLen:81920,81888 
09:07:51 [s3Cloud.cpp ThreadAVS3StreamRecv:396] D> g_nObsVideoSocket=6
09:07:51 [s3Cloud.cpp InitS3AvStream:873] E> p2p start video interface ,Init share mem
09:07:51 [s3Cloud.cpp ThreadAVS3StreamRecv:399] D> g_nObsAudioSocket=7
09:07:51 [AacEnc.cpp InitAACEncoder:25] E> Init AAc encoder:16000,8000,1,16 
09:07:51 [serviceinterface.cpp Call_Service:140] D> ModuleId:1,communicationPort:11000,cmdId:1002
09:07:51 [s3Cloud.cpp InitS3AvStream:898] E> 000 p2p start audio interface ,Init share mem G711A_SHM_BASE_NUM = 50
09:07:51 [main.cpp StartListenAlarmMsg:117] E> alarm port = 37631 
09:07:51 [serviceinterface.cpp Call_Service:140] D> ModuleId:5,communicationPort:15000,cmdId:5015
09:07:52 [main.cpp StartListenAlarmMsg:130] E> start alarm listen socket success 37631 
09:07:52 [AacEnc.cpp InitAACEncoder:81] E> aac encoder inited success 
09:07:52 [s3Cloud.cpp deleteOldFiles:129] E> Enter into delete old files :/var/s3cloud 
09:07:58 [main.cpp ThreadAlarmMsgRecv:89] D> MSG:keepalive ,Msglen:10
09:08:04 [main.cpp ThreadAlarmMsgRecv:89] D> MSG:MB0:0-0-1920-1080-1280-720 ,Msglen:27
09:08:04 [s3Cloud.cpp ThreadAVS3StreamRecv:480] E> 000 Start new event recording :1741507684 
/mnt/mtd/ipc/cfg/colink.conf file not exist 
curl version = 80294 
<HASH_REMOVED>
curl_easy_perform() failed: Couldn't resolve host name
09:08:04 [s3Cloud.cpp ThreadAVS3StreamRecv:485] E> Get upload credentials failed :1741507684 
09:08:04 [s3Cloud.cpp ThreadAVS3StreamRecv:480] E> 000 Start new event recording :1741507684 
/mnt/mtd/ipc/cfg/colink.conf file not exist 
<HASH_REMOVED>
curl_easy_perform() failed: Couldn't resolve host name
09:08:04 [s3Cloud.cpp ThreadAVS3StreamRecv:485] E> Get upload credentials failed :1741507684 
09:08:04 [s3Cloud.cpp ThreadAVS3StreamRecv:480] E> 000 Start new event recording :1741507684 
/mnt/mtd/ipc/cfg/colink.conf file not exist 
<HASH_REMOVED>
curl_easy_perform() failed: Couldn't resolve host name
09:08:04 [s3Cloud.cpp ThreadAVS3StreamRecv:485] E> Get upload credentials failed :1741507684 
09:08:04 [s3Cloud.cpp ThreadAVS3StreamRecv:480] E> 000 Start new event recording :1741507684 
/mnt/mtd/ipc/cfg/colink.conf file not exist 
<HASH_REMOVED>
curl_easy_perform() failed: Couldn't resolve host name
09:08:04 [s3Cloud.cpp ThreadAVS3StreamRecv:485] E> Get upload credentials failed :1741507684 
09:08:05 [s3Cloud.cpp ThreadAVS3StreamRecv:480] E> 000 Start new event recording :1741507685 
/mnt/mtd/ipc/cfg/colink.conf file not exist 
<HASH_REMOVED>
```
Note: At 9:08:04 i moved the camera a bit round. then it started to output with `000 Start new event recording `
probably i need to get the ewelink app to see how local record works with that firmware version?

roleoroleo, 2025-03-09:
I don't know exactly what process is responsible for local recording.
I think you should activate the app and then sniff the internal traffic or the process output.

flreinhard, 2025-03-21:
@roleoroleo how would you do the sniffing? does sonoff-hack ship ready to use tools? I haven't found any.
Did you build tcpdump/tcpflow et.al four your self for development?

roleoroleo, 2025-03-25:
I compiled these:
[tcpdump.gz](https://github.com/user-attachments/files/19455548/tcpdump.gz)
[libpcap.so.1.9.0.gz](https://github.com/user-attachments/files/19455602/libpcap.so.1.9.0.gz)
[strace.gz](https://github.com/user-attachments/files/19455643/strace.gz)

I don't remember if tcpdump is statically linked, so I don't remember if libpcap is needed.

closed 2025-07-24 by github-actions[bot]