roleoroleo/sonoff-hack #194: Cant set timezone and Mqtt is not working

DeadlySin2, 2025-03-21:
MQTT not working 
Timezone can't be changed, I have tried several timezones by i only see HSK-6

[root@Livingroom]# cat /mnt/mtd/ipc/app/version
V5520.2053.0508build20230616

[root@Livingroom]# /mnt/mtd/ipc/app/s3Cloud --help
/mnt/mtd/ipc/cfg/colink.conf file not exist
curl version = 80294
4df076d5dec38a2288730b053c03b49af28d373fa48423e5da6ba9442ade1329
curl_easy_perform() failed: Couldn't resolve host name
/mnt/mtd/ipc/cfg/colink.conf file not exist
dd48e0517f183eb7c31f07aa88451a24e1da4ab35edb58e5b1fc86da15525a28
curl_easy_perform() failed: Couldn't resolve host name
/mnt/mtd/ipc/cfg/colink.conf file not exist
6d6c8087ead46e6377f8d37028b35b9d1054e2856fe5f400f21206b7a229a404
curl_easy_perform() failed: Couldn't resolve host name
/mnt/mtd/ipc/cfg/colink.conf file not exist
8d66216d445911f19df12faf46ce5490f50aaa038f9529b5390e505a41c91afc
curl_easy_perform() failed: Couldn't resolve host name
/mnt/mtd/ipc/cfg/colink.conf file not exist
df80ccd93c7e4236d65f9bc3bede2dcbb3c20fdaa6f38da89d78abb088e84430
curl_easy_perform() failed: Couldn't resolve host name

My be there are any workarounds? firmware 1.7

roleoroleo, 2025-03-25:
These days I'm busy with family issues.
I hope to be able to do a release next weekend.

DeadlySin2, 2025-03-26:
Thanks for your work. 
I have downlded the last release from actions, but im still haveing isuies with the PTZ motion in Home Assistant, probably i do not use the right commands. 
    pan: RIGHT
    distance: 0.05
    move_mode: ContinuousMove
Send camera right but on randome distance. Same for all other commands. 
And i cant use it for motion detection as it is switching on the local record  despite the fact that local record switch is turned off

roleoroleo, 2025-03-29:
> pan: RIGHT
> distance: 0.05
> move_mode: ContinuousMove

Can you send me a screenshot of this panel?

> And i cant use it for motion detection as it is switching on the local record despite the fact that local record switch is turned off

When you enable motion detection, the local record switch change to "on". Right?

DeadlySin2, 2025-03-30:
@roleoroleo 

> When you enable motion detection, the local record switch change to "on". Right?

No, it stays off

roleoroleo, 2025-04-05:
I don't understand the problem, could you explain me better?

DeadlySin2, 2025-04-07:
I do not need local record, but i need motion detection. If i switch motion detection on camera begin to writevideofiles localy so it is inconvinient. 
Local record is turnrd off but the recording is enabled.

[image]

[image]

closed 2025-04-15 by DeadlySin2