roleoroleo/sonoff-hack #195: Wrong time displayed while correct timezone is set

flreinhard, 2025-03-21:
# How can you see the error?
Time in snapshot, rtspstream and on commandline wrong.
In my case it's 15:00:00 in the stream/picture, but in real it's 16:00:00

# requirements
- ntp server is setup and sync is working
- timezone configured (in my case Europe/Brussels)

# what i think is wrong
if you call `date` it will rely on /etc/TZ which is a symlink to /tmp/TZ to make a localtime from UTC. Must be something custom in the glibc that got shipped with the camera.
Now in my case the content is `EST+0:00:00` which is wrong in two different ways:

1.  it does not match my timezone
2. the format is utterly wrong, [gnu.org has the details](https://www.gnu.org/software/libc/manual/html_node/TZ-Variable.html)

Since i live in CET Timezone which is `UTC+1` the content should be `CET-1`
Works for me in theory and real life:
```
root@TestCam]# cat /etc/TZ 
EST+0:00:00
[root@TestCam]# date
Fri Mar 21 21:32:12 EST 2025
[root@TestCam]# date -u
Fri Mar 21 21:32:14 UTC 2025
[root@TestCam]# echo "CET-1" > /etc/TZ 
[root@TestCam]# date
Fri Mar 21 22:33:16 CET 2025
[root@TestCam]# date -u
Fri Mar 21 21:33:19 UTC 2025
```

Is it only me having that issue because i refused to configure the camera with ewelink?
What's your contents of /etc/TZ?

bob-cook, 2025-04-05:
I haven't upgraded to 0.18 yet, so maybe my experience will be different than yours. I live in UTC-8 but found I had to set the timezone to UTC+8 (Asia/Hong_Kong) in order to get the correct local time. There is likely a backwards subtraction going on somewhere, but I've never spent any time trying to find it.

roleoroleo, 2025-04-05:
Please, update to the last version and try again.

bob-cook, 2025-04-05:
Updated to 0.18 this morning, timezone issue (inverted UTC offset) is still there.

closed 2025-08-06 by github-actions[bot]