roleoroleo/sonoff-hack #199: MQTT Error

DeadlySin2, 2025-04-30:
I'm reciving warning from MQTT:
Received message on illegal discovery topic 'homeassistant/binary_sensor/Livingroom_camera_mot ion/config'. The topic contains non allowed characters. For more information see https://www.home-assistant.io/integrations/mqtt/#discovery-topic

Can anyone help to solve the issue. I'm using the latest firmware

roleoroleo, 2025-05-07:
Are you using my integration or the MQTT discrovery feature of the cam?
Is the space char in the topic a typo?

[image]

DeadlySin2, 2025-05-12:
I'm using manual MQTT sensor on adress "sonoffcam/Livingroom/motion_detection" and your integration to get cammera entity. Motion detection of your integration is not working on my system.

DeadlySin2, 2025-05-15:
It helped. Thanks. The only thing that is not working is that i do not recieve "motion_stop" message thougt MQTT

DeadlySin2, 2025-05-20:
It is not generated. 
I have the following settings:
Motion event location =2 
Keepalive =120
Topics Prefix = 
Discovery Prefix = homeassistant
Name = 
Enable MQTT Discovery = 0
QoS = 1
Retain for birth and will messages = 1
Retain for motion messages = 0
Retain for motion image messages = 0

Motion detection = On 
Detection sensitivity = LOW
Local record = OFF
IR led = AUTO

Sensor has the following syntaxis: 
binary_sensor:
  - state_topic: "sonoffcam/Livingroom/motion_detection"
    name: "Livingroom motion"
    device_class: motion
    payload_on: "motion_start"
    payload_off: "motion_stop"
    qos: 1

roleoroleo, 2025-05-24:
Strange.
The stop message is hardcoded and it doesn't depends on the configuration.
I will check it.

roleoroleo, 2025-05-29:
It seems working.
Could you run the process in debug mode?
```
killall mqtt-sonoff
mqtt-sonoff -d
```
and post the output after a motion detection?

- run the process
- create a motion
- wait 15 seconds
- break
- copy/paste the output

roleoroleo, 2025-06-17:
Ok, I need to check the code.

roleoroleo, 2025-07-06:
It's harder than i thought.

roleoroleo, 2025-07-06:
Try this binary:
[mqtt-sonoff.gz](https://github.com/user-attachments/files/21091773/mqtt-sonoff.gz)

DeadlySin2, 2025-07-22:
Everything is working perfectly. Thanks for your support